rst-tu-dortmund/teb_local_planner #6: trajectory optimization is not stable in some cases.

asimay, 2016-05-20:
Hi, your package is a very good solution!
but i come across some issue when doing some tests with default parameters.
I find trajectory optimization is not stable in some cases, especially the obstacle is close to start or end. issues:
1.The trajectory swithes from one to another all the time, and not stable. please see screenshot 1&2.
2. the final trajectory is not the best trajectory from human. please see screenshot 3.
pics show below:
[image]
[image]
[image]

thank you!

asimay, 2016-05-20:
if the obstacles is far away from start or end, the solution acts very well!

croesmann, 2016-05-20:
Hi asimay,

thank you for your feedback and the related issue report.
In my opinion, the *test_optim_node* in combination with interactive markers is quite helpful in order to stress the optimizer to its limits and check how it performs.
Your two issues are both related to infeasibility. Let me explain my thoughts in the following:

First of all, it seems like the minimum distance to obstacles (`min_obstacle_dist`) in your scenario is set to approx. 1.0 meters. Thus the optimizer starts penalizing for distances closer than (`min_obstacle_dist`+`penalty_epsilon`). 

In your screenshot 1 the start pose of the robot is included in the interior of the infeasible area around the rightmost obstacle. The optimizer is therefore not able to find a feasible trajectory (which is correct technically), but however due to the formulation of the optimization problem we even obtain a partial feasible trajectory which is more or less a feature instead of an issue in my opinion, especially in case of the second scenario. However, here with a blocked start, we do not have any advantages of the partially feasible solution. But why is the solution oscillating? Since the start is fixed during optimization and since it is located in the interior, penalty values in the cost function are huge! This implies large gradients and hence larger update steps. In the subsequent step, the optimizer might correct its solution, but again with a large step. In case of multiple alternative candidates this can of course trigger switching between best candidates. However, in such case the planner is not well configured (too large `min_obstacle_dist` value), since this issue implies that the robot physically overlaps with the obstacle.
In case of navigation, the *teb_local_planner* would immediately detect the first infeasible part of the trajectory (a feasibility check is performed before commanding the robot) and consequently the robot will not move until the start is not blocked anymore. 

Despite the oscillations, screenshot 2 indicates an interesting effect at the goal.
The goal is in the interior of the forbidden region around the leftmost obstacle. The optimizer 
shifts the poses (red arrays) along the trajectory (towards the start). This is ok, since no feasible solution exist to reach the goal w.r.t. all constraints. But instead of obtaining a totally infeasible trajectory, the optimizer provides us with a trajectory that is feasible up to the leftmost obstacle is reached (as already noted in the last paragraph). Hence the robot will navigate (as soon as the start is not blocked anymore) until the feasibility check fails which causes the robot to brake. The number of future poses taken into account for the feasibility check is specified with `feasibility_check_no_pose`. It is worth to mention, that if this is not the final goal but a virtual (intermediate) one on the global plan, the infeasibility is often resolved automatically as the robot proceeds moving, due to the moving horizon strategy. Here it is also a big advantage to obtain partially feasible trajectories instead of no solution at all (just think about many (but not all!) constrained optimization techniques that are not able to find a reasonable solution here).

Indeed, by looking at screenshot 3, the obvious solution is missing.
But the result is reasonable:
The exploration algorithm for new candidate trajectories only accepts new trajectories if they are 
not intersecting with any obstacle. Since the goal is still blocked, it won't add any new one. Already found trajectories are kept as long as they fulfill some requirements and the planner will never remove the last candidate.
I am not sure if the current strategy should be relaxed since it is not intuitive to distinguish between multiple infeasible trajectories. In navigation scenarios, the global plan is always injected as candidate besides the exploration of new trajectories. We also assume that the goal is not blocked all the time due to our moving horizon, thus the actual time-optimal trajectory might become revealed during run-time. And if not, it is more desired to follow a previously found (suboptimal) trajectory instead of generating new infeasible ones. However, if you make any experiences in practical navigation scenarios that require a different behavior please let me know :).
By the way, the area of exploring new candidates (which is sample based) can be adjusted by changing the corresponding parameters (but it influences the computation time significantly). The exploration graph can be visualized using `rqt_reconfigure`. But it will not affect anything, if the goal is still blocked.
 

Just for the record:
Despite from the blocked start and blocked goal, further oscillations between multiple alternative trajectories can occur rarely if they are sharing a similar cost value (simple scenario:  one obstacle  located laterally at the middle between start and goal). A probable reason is numerical noise. However, I already added a small hysteresis that can be tuned by parameter `selection_cost_hysteresis`. The default value deactivates the hysteresis. However, I am going to slightly adapt the default value for the next release. A third cause for oscillations that I cannot handle well at the moment are not related to different trajectory candidates but only to a single solution. If the measured robot pose and obstacles are noisy, the optimizer sometimes oscillates between e.g. left-turn parking and right-turn parking or forward/backwards. It occurs rarely for my robot, and usually the solution converges after some oscillations.

asimay, 2016-05-23:
awesome! Thank you for your answer! I really learned a lot from you. Thanks!

closed 2016-05-23 by asimay