rst-tu-dortmund/teb_local_planner #10: Allow setting of minimal velocity

balamesh, 2016-06-08:
Hi,

is it possible to implement a minimal velocity? We are using the teb_local_planner with success on either simulated or real ackermann driven robots.
But in some cases we are facing the problem that the robot base cannot execute the desired velocity.
E.g. it would only start driving at a minimal velocity of 0.1 m/s.

Best regards and thumbs up for the great work!

croesmann, 2016-06-15:
Hi @balamesh,

in general, adding a lower bound on the minimum velocity should be possible. However, integrating this into the optimization scheme significantly increases complexity since we already support switching between forward and backwards motions  (it might introduce non-smooth cost functions or even more local minima).

In my opinion, adjusting the controls/velocities *after* obtaining the optimization result should work fine in many applications, e.g. using a simple three-point-controller before forwarding the velocities to the actual base controller:
 
    if ( abs(v_out >= v_min) ) v_out = v;
    else if ( abs(v_out >= eps) ) v_out = v_min;
    else v_out = 0; 

However, the threshold `eps` depends on
* the maximum acceleration (the robot must be able to reach the subsequent discrete state of the local plan) 
* and the goal tolerance (the robot must be able to reach the desired goal region)

If `v_min` is small and `eps` is chosen appropriately, it should not influence stability significantly to my mind.

It would be really straight forward to add those few lines to the `computeVelocityCommand()` method in `teb_local_planner_ros.cpp` but in my opinion it is more suitable and intuitive if these platform dependent heuristics are included in the robot's base controller as long it is not subject to the actual optimization/planning scheme.

nlimpert, 2016-09-16:
Yes thanks a lot for your help. We managed to solve this by implementing a minimum velocity on the controller side.
And sorry for the late response!

closed 2016-09-16 by nlimpert