rst-tu-dortmund/teb_local_planner #11: wheelbase param and expected location of base_link for carlike robots

beetleskin, 2016-06-15:
The wheelbase parameter name and description are somewhat misleading in my case: We have a car-like robot with steering axle in the front as well as the drive shaft (like a normal car). So steering and drive happen on the front axle. According to the description ..

> The distance between the drive shaft and steering axle. The value might be negative for back-wheeled robots (only required if ~<name>/cmd_angle_instead_rotvel is set to true). 

.. the value should be set to zero in our case, shouldn't it? However, the term _wheelbase_ means the discance between front and rear axle. So which one is meant here, or did I misunderstand the parameter description?

Related: http://answers.ros.org/question/237035/steering-axis-of-carlike-robot-with-teb_local_planner/

croesmann, 2016-06-15:
See my answer [here](http://answers.ros.org/question/237035/steering-axis-of-carlike-robot-with-teb_local_planner/).

Indeed, the wheelbase definition was not correct. I have fixed it on the [wiki page](http://wiki.ros.org/teb_local_planner#Robot_Configuration_Parameters).

Thanks!

beetleskin, 2016-06-16:
Thanks a lot for the detailed answer on answers.ros.org, that solves my problem and clears a lot of questions.

closed 2016-06-16 by beetleskin