rst-tu-dortmund/teb_local_planner #12: Teb Planner claims trajectory not feasible without reasons

schultza, 2016-07-06:
Hi,

I am facing a problem, where the teb_planner claims, that the trajectory which he wants to plan along isn't feasible I can't see any reasons this is the case.
My setup is following:

- Gazebo Simulation
- Local planner: teb_local_planner
- Global planner: sbpl_planner or self written carrot planner (both same problem)

After setting a new goal the teb planner plans along the global plan, which looks good, after driving the base a few meters, it changes in the short horizon mode and keeps resetting itself with following output: 

Debug mode was on:

```
[ INFO] [1467789686.934077314, 1142.800000000]: odom received!
[ WARN] [1467789690.670318292, 1144.963000000]: Map update loop missed its desired rate of 5.0000Hz... the loop actually took 2.5520 seconds
[DEBUG] [1467789694.034116645, 1146.903000000]: initTEBtoGoal(): number of generated samples is less than specified by min_samples. Forcing the insertion of more samples...
[DEBUG] [1467789694.039371530, 1146.903000000]: TebOptimalPlanner::isHorizonReductionAppropriate(): Distance between consecutive poses > 0.9*min_obstacle_dist
[ WARN] [1467789694.039607752, 1146.903000000]: TebLocalPlannerROS: trajectory is not feasible, but the planner suggests a shorter horizon temporary. Complying with its wish for at least 10s...
[ WARN] [1467789701.844726254, 1151.502000000]: TebLocalPlannerROS: trajectory is not feasible. Resetting planner...
[DEBUG] [1467789704.087700070, 1152.800000000]: initTEBtoGoal(): number of generated samples is less than specified by min_samples. Forcing the insertion of more samples...
[ WARN] [1467789704.089528617, 1152.800000000]: TebLocalPlannerROS: trajectory is not feasible. Resetting planner...
[DEBUG] [1467789707.287139842, 1154.699000000]: initTEBtoGoal(): number of generated samples is less than specified by min_samples. Forcing the insertion of more samples...
[ WARN] [1467789707.289574536, 1154.699000000]: TebLocalPlannerROS: trajectory is not feasible. Resetting planner...
[DEBUG] [1467789710.816752726, 1156.699000000]: initTEBtoGoal(): number of generated samples is less than specified by min_samples. Forcing the insertion of more samples...
[ WARN] [1467789710.818900947, 1156.699000000]: TebLocalPlannerROS: trajectory is not feasible. Resetting planner...
```
In this video you see a video of the behavior:

[Video Link Youtube](https://youtu.be/UEoWs_MmJuw)

Here is the [parameter file](http://pastebin.com/KNvmewsr) used for the teb_planner.

beetleskin, 2016-07-06:
Can confirm, it also happens rather often in our setup.

croesmann, 2016-07-11:
Hi,

thank you very much for reporting this issue along with the video!
This issue must definitely be addressed. But I cannot see the actual cause right now.
I haven't seen that in my setup before. I've also applied your parameter configuration on a simulated carlike robot without an issue at first glance (but with the GlobalPlanner).
Unfortunately, I have not much time for investigating things in the code or changing major lines of code during the next 4-5 weeks.

I'll let you know when I have an idea about the cause or a possible fix. If you have more details or observations on that issue (e.g. for helping me in replicating it), please share it here.

schultza, 2016-07-13:
Hi, 
I found a parameter in my costmap_common_params.yaml which maybe was causing this issue.
The parameter I found was `footprint_padding` I dont know exactly what this parameter does (and I neither know why it was in my parameter file :D ) but the value was 1.0.
The parameters seems to be a scaling factor for the actual footprint of the robot, which is already pretty big at our robot (2.5 x 1.5 meters) and it was even expanding the footprint more, so my guess is the planner wasn't able to move along a path because all the time he tried, the footprint was inside some obstacles, while the actual robot wasn't.

I don't know if this is the answer to this problem but it is a step :)

croesmann, 2016-11-15:
Indeed, the 'footprint_padding' is important since it is included in the feasibility check of the planner (robot costmap footprint + padding). I suggest to specify an appropriate costmap footprint and set the padding to zero. 

I revised some parts of the code regarding parameter checking, initialization and backup modes in case infeasible trajectories are detected in the newest version (0.6.5). The code might be tested also in indigo. I want to test the changes for a couple of weeks before I backport the important changes to indigo/jade.

Since there are some other issues that are related to infeasibility, I close this one now. 
We can reopen this issue report if it is not resolved properly. Please comment in that case ;-)

closed 2016-11-15 by croesmann

MohammadDehghanni, 2018-12-23:
I have the same issue. I have combined robot_carlike_in_stage and dyn_obst_costmap_conversion. I mean I've added a dynamic obstacle to the stage(maze_world) and I don't know why the car doesn't take static walls into account and wants to move through them and in many situations I have "TebLocalPlannerROS: trajectory is not feasible. Resetting planner..." message and the car is freezing up.