rst-tu-dortmund/teb_local_planner #13: Free goal yaw

doisyg, 2016-07-11:
Hello,
First of all, thank you very much for your work. The teb_local_planner is by far the most efficient and robust local planner I ve tested in a long time.
I was wondering if there is a way to relax the yaw goal constraint, i.e that the robot just stop on its goal without trying to respect the yaw constraint given by the goal message. Usually I achieve this kind of behavior by setting the yaw tolerance to 2*PI, but here it seems the maximal acceptable value is 2.0 rad.

closed 2016-07-11 by croesmann

croesmann, 2016-07-11:
Hi,

thanks for reporting this and providing me with your feedback!
 I have increased the upper bounds in 08ff168. Changes are also cherry-picked
to indigo and jade.

However, this tolerance is used only for checking if the goal is actually reached.
The optimizer still tries to find a feasible solution to the actual desired orientation but yes, the planner will stop at the goal with any orientation then.
I already wanted to add an option to allow users to relax the orientation even in optimization.
But this requires some changes to the underlying optimization graph which could take some hours. Probably I can find some spare-time after summer (break) in order to add this feature.

doisyg, 2016-07-12:
Hi,
Thanks a lot, tested and working for my purpose. Very curious to see this feature included in the optimization.

commit 08ff1681 (2016-07-11): Enlarged upper bounds on goal position and orientation tolerances
in dynamic_reconfigure. Fixes #13.
commit e60329b8 (2016-07-11): Enlarged upper bounds on goal position and orientation tolerances
in dynamic_reconfigure. Fixes #13.

(cherry picked from commit 08ff1681cfbd4d81a0ca53e909cc87866f6a92a1)
commit 683b9513 (2016-07-11): Enlarged upper bounds on goal position and orientation tolerances
in dynamic_reconfigure. Fixes #13.

(cherry picked from commit 08ff1681cfbd4d81a0ca53e909cc87866f6a92a1)