rst-tu-dortmund/teb_local_planner #14: min. turning radius higher than 5m

glebos, 2016-08-10:
Hello,

is it possible to apply a min. turning radius higher than 5m?

Kind regards.

beetleskin, 2016-08-22:
Speaking of config limits, is there a reason to limit the velocity to 2m/ss?

croesmann, 2016-08-27:
There are no theoretical reasons for the limits. I set them according to my use cases and haven't thought about general cases in the past. 

According to the min turning radius there are just practical limitations:
If the minimum turning radius is substantially high (hence only small steering angles are allowed) the problem size drastically increases: The trajectory length could be huge since the robot must often switch moving directions (alternating forward and backward motions) in order to achieve a desired pose. Despite the fact that the computation time increases, the constraint approximation could be less restrictive and hence leading to solutions that are not feasible for the real robot.

I am going to update the code soon, since currently I do not have spare-time left.
Same for issue #15.

croesmann, 2016-08-31:
I changed the upper bounds in commit 216fe3d.
If you need further adjustments, please let me know.

Note, optimization weights and *dt_ref* as well as *dt_hyst* are not
tuned for velocities and accelerations beyond the default values (e.g. >1 m/s). 
Just increasing the maximum velocity bounds without adjusting the other parameters might lead to an insufficient behavior.

closed 2016-08-31 by croesmann

beetleskin, 2016-09-01:
Thanks a lot for the adjustments. Do you have further suggestions when driving with lets say 20m/s?

glebos, 2016-09-09:
Thank you very much for the adjustments, appreciate it!

commit 216fe3d7 (2016-08-31): Increased bounds of many variables in dynamic_reconfigure. Resolves #14.
The particular variables are maximum velocities, maximum accelerations,
minimum turning radius,...
Note: optimization weights and dt_ref as well as dt_hyst are not
tuned for velocities and accelerations beyond
the default values (e.g. >1 m/s). Just increasing the maximum velocity
bounds without adjusting the other parameters leads to an insufficient behavior.
commit 522a97a3 (2016-12-23): Indigo - Kinetic Synchronization (#34)

Backporting kinetic features to indigo

* Support for omnidirectional drives

* Changed inner velocity storage object to geometry_msgs::Twist to also account for the strafing velocity (and later acceleration). This change caused some function prototype modifications.

* Limiting y-acceleration (strafing acceleration) is now supported if holonomic mode is enabled (vel_max_y > 0)

* Increased bounds of many variables in dynamic_reconfigure. Resolves #14.
The particular variables are maximum velocities, maximum accelerations,
minimum turning radius,...
Note [... cut]