rst-tu-dortmund/teb_local_planner #16: Oscillation in corridors

doisyg, 2016-09-02:
Hello,
When following a straight global path in a narrow corridor (wrt the width of the robot), i have oscillations on the trajectory that i can't explain. These oscillations are not present when navigation in wider spaces and i don't think they are due to localization noise.
Any tips or ideas to narrow the search?

doisyg, 2016-09-02:
I was able to greatly reduce these oscillations by decreasing the inflation_dist parameter (originally 0.5, changed for 0.1). Do you think there is a way to keep the desired behavior of rounding obstacle with a safe margin (the reason for the 0.5 value), without generating trajectory oscillations in a narrow space?

For reference, here are my parameters:

```
recovery_behavior_enabled: False

TebLocalPlannerROS:

 odom_topic: /encoder/odom
 map_frame: /map
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5
    
 # Robot
         
 max_vel_x: 0.42
 max_vel_x_backwards: 0.11
 max_vel_y: 0.0
 max_vel_theta: 0.6
 acc_lim_x: 0.08
 acc_lim_theta: 0.5
 min_turning_radius: 0.0
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
   type: "polygon"
   radius: 0.35 # for type "circular"
   line_start: [-0.3, 0.0] # for type "line"
   line_end: [0.3, 0.0] # for type "line"
   front_offset: 0.2 # for type "two_circles"
   front_radius: 0.2 # for type "two_circles"
   rear_offset: 0.2 # for type "two_circles"
   rear_radius: 0.2 # for type "two_circles"
   vertices: [[-0.27, 0.315], [0.27, 0.315], [0.27, -0.315], [-0.27, -0.315]] # for type "polygon"

 # GoalTolerance
    
 xy_goal_tolerance: 0.3
 yaw_goal_tolerance: 3.2
 free_goal_vel: False
    
 # Obstacles
 
 inflation_dist: 0.1   
 min_obstacle_dist: 0.1
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 2
 no_outer_iterations: 2
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.05
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 100
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet
 alternative_time_cost: False # not in use yet

 # Homotopy Class Planner

 enable_homotopy_class_planning: False
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 1
 selection_cost_hysteresis: 1.0
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False
 
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False
```

dharmp15, 2016-09-10:
1. Lower the angular acceleration rate , max_vel_theta: 0.6 (check with 0.01 to 0.3 ).
2. increase the max_global_plan_lookahead_dist: 3.0 (check with 6 to 8 range)

doisyg, 2016-09-12:
The robot is not overshooting in steering when no obstacles is in range. I don't think it has to do with the max_vel_theta value, but I will conduct more tests.

AchmadFathoni, 2018-06-29:
How is it going?