rst-tu-dortmund/teb_local_planner #17: Vehicle unable to trace the planned local path

nitin5, 2016-09-08:
Greetings,
I am attempting to run teb-local-planner on a bigger vehicle. The size of wheel base is 1.9 meters and turning radius is appx. 4.5 meters. I have observed below mentioned undesired behaviours by the planner. I request you to suggest possible remedy for these behaviours.

I  have observed that local planner is unable to trace the planned path. It prefers to take a reverse and align vehicle with the global plan but is unable to execute the plan. Local plan continuously changes. Sometimes, local planned path moves through the walls.
When a global planner plans a straight line path, robot is moves in snake like fashion. This is more prominent at lower velocities.

It is also observed that acceleration limits on rotation velocities has less effect. Robot is unable to achieve max speed if dt_ref is small. 

I am attaching configuration files used for stage simulation. I am using pre-compiled teb on ROS Kinetic for Ubuntu 16.04.

Thanks,
Nitin
[teb_local_planner_launch-20160908T182842Z.zip](https://github.com/rst-tu-dortmund/teb_local_planner/files/462401/teb_local_planner_launch-20160908T182842Z.zip)

dharmp15, 2016-09-10:
**1. In teb_local_planner_params.yaml: cmd_angle_instead_rotvel: false : set it "true" // Because if it false stage simulator takes the rotation angle instead of the rotvel (w=v/r, if v is non zero; if v=0, w=0) (twist msgs)
2. In car_robot.inc:  odom_error [0.03 0.03 999999 999999 999999 0.02] from where you taken this value?? For simulation, set odom error zero [0 0 0 0] or take localisation by 'gps' that will not give any odom_error.
http://playerstage.sourceforge.net/doc/stage-cvs/group__model__position.html
3. In amcl_param: odom_model_type: "diff" set it "car"
4. In robot_carlike_in_stage.launch  set  this param (<param name="/use_sim_time" value="true"/>) = false.
5. Check Homotopy Class Planner parameters also for optimisation. (if not possible to optimise set it false).**

closed 2016-09-16 by nitin5

nitin5, 2016-09-16:
Hello,

> In robot_carlike_in_stage.launch set this param use_sim_time value= false.
This will lead to time mismatch error in simulation.

> [ WARN] [1474045897.900828321]: Costmap2DROS transform timeout. Current time: 1474045897.9008, global_pose stamp: 3.7000, tolerance: 0.5000