rst-tu-dortmund/teb_local_planner #18: Add steering speed limit for car-like robots

grieneis, 2016-09-14:
I am trying to use this planner for a 'car-like' robot.  On the robot, the rate of change of the steering wheel is limited, i.e. it can only change the steering angle at ~1 rad/s.  I want to add this constraint to the planner to generate paths that are feasible.  Do you have any suggestions about how to implement this constraint?

croesmann, 2016-09-22:
Hi,

since I have updated the underlying optimization graph structure recently (commit f8f26c70ea9e201), I also added the steering rate limitation according to your request. You can try it out if you check out the *steering_rate* branch. You might want to inspect the changes.

Currently, I am setting the steering angle to zero if the linear velocity is zero as well.
Another strategy would be to extend the optimization problem to keep the previous steering angle if v==0, but this would result in a more complex structure (edges also depend on the predecessor turning radius). However, if there are any issues with the current strategy, we could integrate the latter.

If you have any feedback or test results, please let me know.

Do not forget to set the `wheelbase` parameter.

Cheers, 
Christoph

PS @nitin5 and @doisyg: I am going to address your issues soon.

bochen87, 2018-03-12:
Hi Christopher,

I just tested the steering_rate branch and I think it could be a nice feature to keep the previous steering angle instead of setting it to 0. Reason: In reality, the steering angle of the wheels do not return immediately back to 0 as it takes time to do so.

Regards,
Bo