rst-tu-dortmund/teb_local_planner #19: too much prejudice towards backward movement in local plan

nitin5, 2016-09-16:
Greetings,
I have observed that teb have preference to turn back and move rather than moving forward as human do.
I am attaching two screen shots to help me pose this question.  I have also observed that even in case of diff-drive robot vehicle do not considers rotation in place as one of the feasible options.
Kindly advise which parameters I should tune so that preference to forward movement is higher.

I am using following parameters for teb

> TebLocalPlannerROS:
> 
>  odom_topic: odom
>     
>  # Trajectory
>  cmd_angle_instead_rotvel: false    #NITIN
>  teb_autosize: True
>  dt_ref: 0.3
>  dt_hysteresis: 0.1
>  global_plan_overwrite_orientation: True
>  max_global_plan_lookahead_dist: 3.0
>  feasibility_check_no_poses: 5
>     
>  # Robot
>          
>  max_vel_x: 0.4
>  max_vel_x_backwards: 0.2
>  max_vel_y: 0.0
>  max_vel_theta: 0.3
>  acc_lim_x: 0.5
>  acc_lim_theta: 0.5
>  min_turning_radius: 0.0 # diff-drive robot (can turn on place!)
> 
>  footprint_model:
>    type: "point"
> 
>  # GoalTolerance
>     
>  xy_goal_tolerance: 0.2
>  yaw_goal_tolerance: 0.1
>  free_goal_vel: False
>     
>  # Obstacles
>     
>  min_obstacle_dist: 0.25 # This value must also include our robot radius, since footprint_model is set to "point".
>  include_costmap_obstacles: True
>  costmap_obstacles_behind_robot_dist: 1.0
>  obstacle_poses_affected: 30
>  costmap_converter_plugin: ""
>  costmap_converter_spin_thread: True
>  costmap_converter_rate: 5
> 
>  # Optimization
>     
>  no_inner_iterations: 5
>  no_outer_iterations: 4
>  optimization_activate: True
>  optimization_verbose: False
>  penalty_epsilon: 0.1
>  weight_max_vel_x: 2
>  weight_max_vel_theta: 1
>  weight_acc_lim_x: 1
>  weight_acc_lim_theta: 1
>  weight_kinematics_nh: 1000
>  weight_kinematics_forward_drive: 1
>  weight_kinematics_turning_radius: 1
>  weight_optimaltime: 1
>  weight_obstacle: 50
>  weight_dynamic_obstacle: 10 # not in use yet
> 
>  # Homotopy Class Planner
> 
>  enable_homotopy_class_planning: True
>  enable_multithreading: True
>  simple_exploration: False
>  max_number_classes: 4
>  selection_cost_hysteresis: 1.0
>  selection_obst_cost_scale: 1.0
>  selection_alternative_time_cost: False
>  
>  roadmap_graph_no_samples: 15
>  roadmap_graph_area_width: 5
>  h_signature_prescaler: 0.5
>  h_signature_threshold: 0.1
>  obstacle_keypoint_offset: 0.1
>  obstacle_heading_threshold: 0.45
>  visualize_hc_graph: False
> 

[image]
[image]

hapi007, 2016-10-20:
i suggest you could tune weight_kinematics_forward_drive to 1000

croesmann, 2016-11-15:
The planner produces time-optimal trajectories. 
Depending on the limits on linear velocity, turn rate and accelerations it is often slower to rotate in place. Obviously, if the optimizer would minimize path length rather than time, rotations in place would be preferred (_this is currently not implemented, but this feature might be added easily_).

As @hapi007 suggested, you can try to heavily penalize backwards motions. Currently it is not possible to guarantee forward motions only in all situations. I am going to add it to my TODO list ;-)

I close this issue now, but you can still continue commenting.

closed 2016-11-15 by croesmann

jubeira, 2018-06-06:
Hello! First of all, thanks for this awesome planner, and the hints in this thread.
I'm using TEB with a configuration that is optimized for path following, which works quite well. In my case, I have a diff drive robot, and I would be interested in implementing a way to minimize path length, and therefore prioritize in-place rotations over backwards movements. Is this still in the TODO list? May I have a pointer to at least check how much effort it would take to implement it?

I've already tried penalizing backwards motions, but it still feels like with rotations in place it will work better (the robot sometimes get's stuck when trying to reach rotate-only goals).

croesmann, 2018-06-07:
Hey! Thanks for your feedback. I played with the minimization of the path lengths in the early development stage of the planner, but personally, I preferred the minimum time solution over the shortest path. Nevertheless, I can fully understand the use-case ;-)

If you want to modify the optimization problem, you need some basic knowledge of the g2o-framework / hyper-graph least-squares-optimization.
Then, it is straightforward to add an optimization edge that minimizes path length (just add a binary edge that minimizes the Euclidean distance between two consecutive poses.
You need to turn of the time-optimal edge of course.

But now, you have to provide additional information to the optimizer, otherwise, the optimization problem is ill-posed:
Since the time-differences (Delta T_k) between two consecutive poses are still subject to optimization, you need to add more constraints:
E.g. you could try to fix the velocity always to its maximum speed or maximize it within its bounds. Or you could try to fix the Delta T vertices during optimization. But in that case, the automatic insertion and deletion of discrete poses does not work anymore since it is based on Delta T. You would need to define a distance metric in SE1, in order to add or remove poses (you could also, of course, try to keep the number of poses fixed at all.)