rst-tu-dortmund/teb_local_planner #22: Trajectory is not feasible

jiangxinyu, 2016-09-23:
I have the same problem as the video: https://www.youtube.com/watch?v=L4Wvehzx818 , I get a Error message that Trajectory is not feasible when I set a goal to the robot. How can I slove this problem? Please help me.

croesmann, 2016-09-23:
The particular problem in the video as been resolved (see #3 and https://www.youtube.com/watch?v=yQJFoCzNoAk).

Without more details on your problem, I cannot assist you with a possible solution.

jiangxinyu, 2016-09-23:
OK, I have changed the costmap_obstacles_behind_robot_dist parameters and it works. But there is another problem on my turtlebot: When I set a goal to my turtlebot, it runs not very continuously, often stop frequently and send a low velocity and rotation to the robot. What will cause this problem?

jiangxinyu, 2016-09-23:
https://www.youtube.com/watch?v=dj2eJVEcIXk
https://www.youtube.com/watch?v=crh1TnTc6tM
This is the video that has this problem.

jiangxinyu, 2016-09-23:
And my teb_local_planner yaml file:
TebLocalPlannerROS:

 odom_topic: odom
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5
    
 # Robot
         
 max_vel_x: 0.4
 max_vel_x_backwards: 0.2
 max_vel_y: 0.0
 max_vel_theta: 0.3
 acc_lim_x: 0.5
 acc_lim_theta: 0.5
 min_turning_radius: 0.0 # diff-drive robot (can turn on place!)

 footprint_model:
   type: "circular"
   radius: 0.2

 # GoalTolerance
    
 xy_goal_tolerance: 0.2
 yaw_goal_tolerance: 0.1
 free_goal_vel: False
    
 # Obstacles
    
 min_obstacle_dist: 0.25 # This value must also include our robot radius, since footprint_model is set to "point".
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet

 # Homotopy Class Planner

 enable_homotopy_class_planning: True
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 4
 selection_cost_hysteresis: 1.0
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False
 viapoints_all_candidates: false

 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False

nitin5, 2016-10-08:
@jiangxinyu 
I am also facing similar problem. Robot gets stuck at tight places and after that planner only generates very small velocity commands. The behaviour is very similar to the videos shown in youtube links. 
Also  planner plan non-intuitive paths. For example
[image]

Have you made any progress?

dharmp15, 2016-10-14:
@jiangxinyu , I also faced the same problem and it resolved by disabling the optimization. Disabling the homotopy class may help you. 
enable_homotopy_class_planning: false

jiangxinyu, 2016-10-15:
@dharmp15 It works, thank you very much!