rst-tu-dortmund/teb_local_planner #23: Costmap clearance issue

madan96, 2016-09-29:
I am trying to use the TEB local planner plugin in the move_base ROS package. I am facing some problem with the costmap formed. Whenever someone/thing moves in front of the LiDAR an obstacle layer is formed as visible in Rviz. 

Now, the problem is that after the person/thing has shifted from its previous position, the costmap doesn't clear the old layer unless there isn't any other obstacle behind that layer. Can you please help and resolve this issue.

nitin5, 2016-09-30:
Please make launch file and test bag available.

madan96, 2016-10-02:
`<?xml version="1.0"?>
<launch>
       <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
      
      
    <param name="base_global_planner" value="navfn/NavfnROS"/>
    <param name="base_local_planner" value="teb_local_planner/TebLocalPlannerROS" />
    
      
    <param name="footprint_padding" value="0.01" />
    <param name="controller_frequency" value="10.0" />
    <param name="controller_patience" value="100.0" />
    <param name="planner_frequency" value="1.0" />
    <param name="planner_patience" value="5.0" />
    <rosparam file="$(find eklavya_navigation)/config/global_costmap_params.yaml" command="load" />
    <rosparam file="$(find eklavya_navigation)/config/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find eklavya_navigation)/config/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find eklavya_navigation)/config/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find eklavya_navigation)/config/nomap_global_costmap_params.yaml" command="load" />
    <rosparam file="$(find eklavya_navigation)/config/teb_local_planner_params.yaml" command="load" />
    <rosparam file="$(find eklavya_navigation)/config/costmap_conversion_params.yaml" command="load" /> 
    
    <rosparam file="$(find eklavya_navigation)/config/base_local_planner_params.yaml" command="load" />
    
    

  </node>

<node pkg="tf" type="static_transform_publisher" name="laser_broadcaster" args="1.15 0 0 0 0 0 1 base_link laser 100" />

	  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find teb_local_planner_tutorials)/cfg/rviz_navigation.rviz"/>

</launch>`

madan96, 2016-10-02:
https://drive.google.com/file/d/0B_FrJjhTmwq_NFBPS2RkNmY5aHM/view?usp=sharing
This is the link to the launch file.
I will upload the bag file and comment the link. In the mean time you can look through the launch file for any errors that I might have made.

nitin5, 2016-10-04:
Please add associated parameter files as well. At first look, it appear you are not refreshing costmaps at desired rate.

croesmann, 2016-11-16:
The issue is related to the configuration of the [costmap_2d](http://wiki.ros.org/costmap_2d). This planner just uses the costmap_2d as part of the navigation stack.

Available parameters to configure the obstacle layer including clearing/marking are listed [here](http://wiki.ros.org/costmap_2d/hydro/obstacles).

If you have any questions regarding costmap_2d, please refer to [ROS Answers](http://answers.ros.org) 
or in case of issues with the costmap refer to https://github.com/ros-planning/navigation.

closed 2016-11-16 by croesmann