rst-tu-dortmund/teb_local_planner #27: teb planner can not compute Velocity Commands in time

hapi007, 2016-10-12:
When i use teb_local_planner in kobuki , it can not  compute Velocity Commands to move_base node in time ,so move_base node print these info always:
[ WARN] [1476253731.838476254]: Control loop missed its desired rate of 3.0000Hz... the loop actually took 3.9523 seconds
[ WARN] [1476253994.264856069]: Map update loop missed its desired rate of 3.0000Hz... the loop actually took 12.3793 seconds

My params is default. When i used dwa, it is all right.
By the way, the cpu top info is fined ,there is 80% idle.
[teb_local_planner_params.txt](https://github.com/rst-tu-dortmund/teb_local_planner/files/523663/teb_local_planner_params.txt)

alexbaucom17, 2017-03-23:
I had this same problem and turning enable_homotopy_class_planning to false fixed the issue for me. My suspicion is that move_base is not multi-threaded and so when teb_planner was taking up all the CPU on the same core as move_base it was causing problems. This would still make it look like there is lots of CPU left but that high idle is all from other cores. I could be wrong about that, but I would try turning off the homotopy planning and see if that works for you.

chenxianbo, 2018-08-15:
i have the same problem, the param is default, when i set the new goal, someone of cpu cores goto 100%

chenxianbo, 2018-08-21:
@hapi007 
maybe your obstacles is too many, I choose polygon costmap_converter,my 'costmap_converter_params.yamls' showed below
----------------------------------------------------------------------------------------------------------------------------------
TebLocalPlannerROS:

  ## Costmap converter plugin   
  costmap_converter_plugin: "costmap_converter::CostmapToPolygonsDBSMCCH"
  #costmap_converter_plugin: "costmap_converter::CostmapToLinesDBSRANSAC"
  #costmap_converter_plugin: "costmap_converter::CostmapToLinesDBSMCCH"
  #costmap_converter_plugin: "costmap_converter::CostmapToPolygonsDBSConcaveHull"
  costmap_converter_spin_thread: True
  costmap_converter_rate: 5
 
 
  ## Configure plugins (namespace move_base/costmap_to_lines or move_base/costmap_to_polygons)
  ## costmap_converter/CostmapToLinesDBSRANSAC, costmap_converter/CostmapToLinesDBSMCCH, costmap_converter/CostmapToPolygonsDBSMCCH
  #costmap_converter/CostmapToLinesDBSRANSAC:
    #cluster_max_distance: 0.4
    #cluster_min_pts: 2
    #ransac_inlier_distance: 0.15
    #ransac_min_inliers: 10
    #ransac_no_iterations: 1500
    #ransac_remainig_outliers: 3
    #ransac_convert_outlier_pts: True
    #ransac_filter_remaining_outlier_pts: False
  costmap_converter/CostmapToPolygonsDBSMCCH:
   cluster_max_distance: 0.5
   cluster_min_pts: 2
   cluster_max_pts: 10000
convex_hull_min_pt_separation: 0.1

closed 2019-04-10 by croesmann