rst-tu-dortmund/teb_local_planner #28: Problem moving differential drive robot through narrow corridors.

jmudit19, 2016-10-12:
Hello sir,
I am trying to use teb_local_planner for passing though narrow paths efficiently. The planner works fine for broad pathways but when I am trying to pass a Robot: 
  type : "line"
   line_start: [-0.42, -0.21] 
   line_end: [0.42, -0.21]
through a door with width ~0.7m. The robot gets stuck to this:
[image]
I am currently using ros-kinetic on ubuntu 15.10.
The following are the configurations used : 

[image]
[image]

And sometimes at this (I cannot increase the inflation radius further because it does not allow the robot to move through narrow pathways ) :
[image]
These are the configurations for global and local planner:
[image]
[image]

I have also tried the same with SBPL lattice planner. Please help!

croesmann, 2016-10-12:
The error message "trajectory is not feasible" indicates that it seems like the robot-costmap-footprint (blue rectangular, defined in the costmap param file) collides with some of the lethal-obstacles cellls slightly at some point on the trajectory.

I am going to revise the planner completely in the next three weeks since I hadn't much resources during the last weeks to investigate the recently reported issues. I'll check this one as well.

nitin5, 2016-10-12:
Dear Chris,
Please also investigate why TEB do not prefer a robot to move backward in straight line as one of the option. In narrow path ways, TEB ignores this option.
Also, I have observed that TEB oscillates between local goal points as selected from the global plan.
Thanks

croesmann, 2016-11-15:
I pushed some significant changes to kinetic-devel (see changelog of version 0.6.5). I revised some parts of the code regarding parameter checking, backwards-initialization and backup strategies.

Please let me know if the proposed changes resolve some of your issues. 
The kinetic-devel branch might also be compiled with indigo/jade.
I want to test the changes for a couple of weeks before I officially backport the important changes to indigo-devel/jade-devel.

andrew8liang, 2016-12-15:
Hi, I also have the same question about moving the robot backward in a straight line on `0.6.5`. 

The expected behavior is: drive backwards. The actual behavior is: initially drive backwards, then rotate 180 degrees to drive forward until we are close to the final goal, then rotate another 180 degrees so that we are facing the correct direction.  This gives us 360 degrees of unnecessary rotation. Is it possible to have the planner put equal weight on forward and backward motion?

My parameters, on 0.6.5:
`allow_init_with_backwards_motion: true`, `weight_kinematics_forward_drive: 0`

Thanks!

croesmann, 2016-12-16:
@andrew8liang Please avoid [cross-posting](http://answers.ros.org/question/250138/differential-drive-favor-backwards-movement-teb_local_planner/) if possible. I usually get notifications for both platforms ;-)
I have answered your question at [ROS answers](http://answers.ros.org/question/250138/differential-drive-favor-backwards-movement-teb_local_planner/). 

In general, questions regarding the planner should be asked at ROS answers.
Bug reports, Pull requests and enhancement requests here.

DPatricia, 2018-02-12:
Hello,
I'm having a similar issue with the robot getting stuck trying to go through a door.
Are there some parameters which I can adjust to manage this?
Thanks

subodh-malgonde, 2018-11-13:
@DPatricia Were you able to figure out a solution for your problem? I have a similar issue. I have a car-like robot (width 0.3 m). No matter how I tune my minimum  obstacle distance and inflation radius, it has a difficulty in entering a narrow door which is 1 m wide.

corot, 2019-04-23:
I just discovered a magic parameter that makes wonders for me: _penalty_epsilon_
With default (0.1):
[image]
With 0.05:
[image]

@croesmann, would be great if you can provide some clue about how this parameter so greatly impacts the behavior of the robot

rrdpereira, 2022-10-04:
> I just discovered a magic parameter that makes wonders for me: _penalty_epsilon_ With default (0.1): [image] [ [image] [...]

Could you share the COMMON, LOCAL, GLOBAL and TEB configuration files? Or just the GitHub Link?