rst-tu-dortmund/teb_local_planner #38: Local Plan problem

jiangxinyu, 2017-01-21:
Hi,
Thank your for the update of the algorithm. But I find a new problem in the latest version of the Kinetic. When the planner sometimes replan the local plan, the plan changes into a mass(see the following picture)
Please help me how to solve this problem?
Thank you very much.
[image]

boilerbots, 2017-02-09:
I have seen this same issue. It starts to happen after reducing the min_obstacle_dist=0.1 and inflation_dist=0.2 The local path will produce the effect that you are showing frequently and typically will not make progress as it approaches an obstacle.

jiangxinyu, 2017-02-09:
@boilerbots so the only way to slove this problem is to increse the min_obstacle_dist or chage the inflation_dist parameters?

boilerbots, 2017-02-09:
I don't have a good solution yet, sorry. I am just commenting to let the maintainer know that there are issues.

croesmann, 2017-02-13:
thanks for confirming @boilerbots. I will start investigating this in 1-2 weeks.

croesmann, 2017-03-13:
sorry, due to some more workload I had to further postpone working on the planner. Starting from next week, I will definitely have some spare time to investigate current issues.

bailizheng, 2019-05-21:
Hello, I am currently experiencing this problem. How do you solve it? Can you talk about the solution? thank you very much!

Noah-Lau, 2023-09-05:
Have this problem be solved ?