rst-tu-dortmund/teb_local_planner #41: Why does TEB plan cross over obstacle inflations

iamkevinzhao, 2017-03-14:
Hi Christoph,

First of all, thanks for contributing this awesome project.

However, something confuses me. I noticed that the produced TEB local plan sometimes crossed over the inflation margin of detected obstacles, which could cause collisions. Is that normal?

[image]

Line in pink in above screenshot was the problem TEB local plan

Look forward to your reply.

Kevin

croesmann, 2017-03-23:
Hi Kevin,

thank you for reporting this issue.
The common inflation layer of the costmap_2d is not taken into account by the planner,
since the internal optimization cannot be applied to grid cells. As an alternative, I added a the possibility to specify a small penalty for robot positions close to obstacles (parameter `inflation_dist`), but I haven't played a lot with it yet.

Nevertheless, your resulting trajectory in pink das not look well. Do you have further information on how I can reproduce this? By the way, your costmap does not seem to correspond to your static map or amcl failed to localize?

tolster710, 2017-12-27:
Hi there! Has there been any progress made on this front?
I have a pretty wide robot that is convinced it can both: fit through tight spaces, and muscle objects out of the way. The desired functionality would be to fail and halt the robot.

Here is a link of a screen cast of the planner operation when an object is introduced. we are using the ```inflation_dist``` parameter. I'm happy to provide any other details which could be helpful in diagnosing.
[gDrive Link](https://drive.google.com/open?id=17fab5SPGthJHOyTEeI45MC2JdALBi_Xt)

marshalshi, 2018-11-12:
Currently I'm using teb for local path planner with `static_layer` costmap, where existing lethal obstacle (cost >100) and non-lethal obstacle (cost < 100). And also, `static_layer` set `trinary_costmap` to `false`for both global planner and local planner such that all full spectrum of intermediate values are used.

Thanks for making this nice project. And I am new for TEB. 

When I tried to move cross `non-lethal` obstacle: the global path plan made a path which was round the `non-lethal` obstacle, but when movebase moving, TEB would cross the `non-lethal` obstacle (which just like the `non-lethal` obstacle didn't exist). 

I am not sure that: Does TEB have the ability to walk round the `non-lethal` obstacle according to cost map value (if existing path which has lower cost sum)? 

[image]
At above example image, the "red" area has <100 cost value. Blue line is global path. Red arrows are TEB path. 

Thanks for helping.