rst-tu-dortmund/teb_local_planner #45: Is it possible to limit the max. feasibility check distance?

luator, 2017-05-16:
Hello,

as already posted on [ROS Answers](http://answers.ros.org/question/261834/what-are-reasons-for-teb-planners-feasibility-check-to-fail/), I have the problem that the feasibility check regularly fails without any obstacles near by. I think the reason is, that my local costmap is too small so that the footprint is outside of it for poses further away from the robot.

Of course an obvious solution is to increase the costmap size or reduce `max_global_plan_lookahead_dist`. But none of these give me a guarantee that the problem cannot occur. Is there some parameter with which I can restrict the maximum distance of poses to the current robot position, when doing the feasibility check (so that it will rather consider less poses for the check instead of moving outside of the costmap)?

croesmann, 2017-05-19:
Hi,

thanks for posting this issue.
I thought that `feasibility_check_no_poses = 0` restricts the maximum distance of poses to just the current robot position. But I wasn't aware of the issue with a too small local costmap.
We could probably add a workaround in  [optimal_planner.cpp:1175](https://github.com/rst-tu-dortmund/teb_local_planner/blob/kinetic-devel/src/optimal_planner.cpp#L1175) to return always _feasible_ for negative `feasibility_check_no_poses`. 
But we should better figure out why footprintCost() of the costmap_2d model detects a collision in your particular scenario.

luator, 2017-05-19:
Hello,

there should be no need to disable the feasibility check completely (at least the current robot pose should always be on the local costmap, otherwise the costmap is kind of useless). I was more thinking about something like `feasibility_check_lookahead_distance` and only check for poses that do not exceed this distance to the robot. One could then set this value based on costmap and footprint size to ensure that the check does not exceed the costmap bounds. So if possible, `feasibility_check_no_poses` poses would be checked, but it would decrease the number when poses are getting too far away.

Or is it maybe already possible with existing parameters to ensure that the n-th pose of the plan is not further than a specific distance away from the robot? I think this would already be enough.

In my particular scenario the problem is that the `costmap_.worldToMap()` inside `footprintCost()` fails, when a point of the footprint is outside of the costmap bounds. The returned cost is then the same as if there would be an obstacle inside the footprint, so from outside this error is unfortunately not detectable.

Best, Felix

luator, 2017-05-19:
This is not a critical issue, as I can always ensure failure-free checks by setting `feasibility_check_no_poses` to a small enough value (worst case to 0). I just thought that it would be nice to be able to look a bit more further ahead without risking to get false negatives due to this issue.

amakarow, 2020-10-17:
Closed as specifying the distance is now possible with #231.

closed 2020-10-17 by amakarow