rst-tu-dortmund/teb_local_planner #47: parallel parking

Benjaminmar8, 2017-07-03:
Hi, All,

We are trying to use teb_local_planner for parallel parking application, anyone has tried this before?  What would be the good set of parameters to start with? Thank you very much.

Regards
Ben

GuillaumeHauss, 2018-03-20:
Kindly interested in that matter too. I'm rather targeting perpendicular parking, but I guess the idea stays the same. So far, I'm struggling with preventing teb_planner from asking the car to switch motion directions like 10 times on a 3m long path..

croesmann, 2018-06-04:
@GuillaumeHauss I fixed a crucial bug today which led to a huge difference between the planned and actually controlled trajectory and hence introducing lots of motion traversals. I would be happy, if you can test this again and let me know if the performance is better :)

GuillaumeHauss, 2018-06-04:
Hi @croesmann,

Thanks for your continuous work on that, then!
Is the new code on a specific branch in the repo, or can I jush pull from the master branch?

croesmann, 2018-06-04:
just pull from the default branch. It's kinetic-devel. I haven't cherry-picked it yet for other distros.