rst-tu-dortmund/teb_local_planner #50: Oscillation in movement

David1Zhang, 2017-07-25:
Hi,
I am trying this local planner in auto-parking with a electric car. Now, I can successfully complete the auto-parking scenario in the simulation.  But the problem is that  the "/cmd_vel/angular/z" is frequently changed within a short time when the car is moving. This effect can be ignored in the simulation but when testing in the car,the controller steer can't adjust the route with a high frequently changeable direction. Especially, when the car is moving near some obstacles, the angular/z is more unstable. 
 So,is there any parameters can be tuned to make the angular/z change more stably? Or other ideas can solve this problem?  

Thanks
[image]

nlimpert, 2017-08-24:
Hi,

for velocity smoothing I use the yocs_velocity_smoother as found in [1]. You should make sure that at least the speed limits are the same like setup in the teb_local_planner. The acceleration limits should not differ too much from those setup in the teb_local_planner.

1: https://github.com/yujinrobot/yujin_ocs

David1Zhang, 2017-09-08:
Thank you for your comment:)

croesmann, 2018-06-05:
Is this problem solved? (Do you use `/cmd_vel/angular/z` as angular velocity or is the steering-angle mode enabled, i.e., that `/cmd_vel/angular/z` defines the angle of the steering wheel?)

If the problem is not solved yet, please try the most recent version of the planner and reopen the issue.

closed 2018-06-05 by croesmann