rst-tu-dortmund/teb_local_planner #51: backwards motion is really aggressive

dkati, 2017-08-24:
Good evening and thanks for all!.
i set my planner as teblocalplanner and i realized that when the direction where the robot has to go,is backwards,the turning is strange...it may get lost...it may go fully backwards(which i dont want to)...it spins fast....i did not touch the parametres at all.my global planner is the the "stock" one of ROS

all i want to do is : 
Force the robot to spin on the same point when it is for backward movement...make a rotation with pi rad and move forward...
by the way my robot is differential drive

thanks

nlimpert, 2017-08-25:
Hi,

short question; what happens if you set the parameter "max_vel_x_backwards" to 0.0?

Regards

dkati, 2017-08-25:
hi @nlimpert and thanks for reply.

the minimum allowed value 0.01 which is already set

nlimpert, 2017-08-25:
Hi,

so I guess you already tried to increase the parameter weight_kinematics_forward_drive?

dkati, 2017-08-28:
@nlimpert that was what i was looking for mister :)
Thanks for that....!solved my issue...
now my last issue with teb....

the dynamic obstacle avoidance is strange.
Robot can recognize the new obstacle but it goes like this:
Moving a bit forward....stop and wait...Moving a bit forward....stop and wait...finally it makes a complete turn in front of the obstacle...the local planner (red) line of rviz is going crazy around obstacle but in the end there is no motion.it stands there forever

croesmann, 2018-06-04:
I close this for now since the original issue has been resolved. @dkati if you observe the second issue even with the newest planner version (kinetic-devel branch), please let me know and open a new issue. Thank you!

closed 2018-06-04 by croesmann

Bhanu5302, 2023-01-18:
In real, my robot is taking taking axial turns to come back its previous position but I want to move my robot to backward also. Now i'm using teb local planner