rst-tu-dortmund/teb_local_planner #52: Positive theta_max is never reached

luke321, 2017-09-04:
I found an interesting problem when using the teb_local_planner for my omnidirectional robot.
I downloaded the planner from source and the problem also shows in the teb_local_planner robot_omnidir_in_stage.launch from the tutorial package.

When making a counter clockwise in place rotation, the maximum velocity is never reached.
Typically the theta veloctiy flucktuates at about 0.2.
When doing clockwise rotation the maximum velocity is reached, for example -0.3 in the tutorial example.

The problem only occurs when turning counter clockwise from standstill, if you start a rotation in clockwise direcation and than switch to counter clockwise, the max velocity is reached.

Here is a video showcasing the problem: [video](https://youtu.be/nmwLAk_W8Cc)

procopiostein, 2018-03-22:
I am facing the same problem...
It seems that after navigating around for a while, the planner starts to reach the full speed while rotating anticlockwise.

croesmann, 2018-06-04:
This could be related to a crucial bug that I fixed today in the kinetic-devel branch. The planner accidently preferred a clockwise turn which was intendend only for some oscillation recovery. As a result, the planner penalized all anti-clockwise rotations and I guess the maximum rotational velocity could not be reached. This affected also the complete control performance. The navigation should be much better now with all robot types. Please check again and feel free to comment on this issue. I close this for now.

closed 2018-06-04 by croesmann