rst-tu-dortmund/teb_local_planner #53: Pause re-planning for x seconds when dynamic obstacle detected

bochen87, 2017-09-18:
Dear Christoph,

I was thinking of implementing a feature that would allow for the following feature:
Dynamic Obstacle get's in the path of the robot. Robot slows down and ultimately stops and waits for X seconds. After x seconds, if the obstacle hasn't moved away, the robot replans the path. 

So essentially instead of an immediate re-planning, I would like to add some lag, since in many instances, the dynamic obstacle will move out of the way by itself and a dynamic re-planning with the other dynamic obstacle also re-planning course will lead to many re-plannings...

What are your thoughts on this? Where and how would you start with such a change?

closed 2017-11-09 by bochen87

zhafree, 2018-02-22:
Hi, I do not think the new version of teb_local_planner can solve the issue you given；Do you Continue to follow up on this issue?

aditya7robo, 2023-12-14:
Hi, has this feature been added/updated? How do I implement this?