rst-tu-dortmund/teb_local_planner #55: Teb planner without a map

MomoTavas, 2017-10-06:
Hi again @croesmann
I have tried to create an empty world and modified the costmap accordingly (I think).  However I am not able to use the 2D Nav Goal in Rviz to get it running. may you take a look and tell me what is wrong?
First I tried with no map at all, but apparantly /map topic is expected. Therefore I imported an empty while floor plan as a map. Can that be the issue?

NODES
  /
    amcl (amcl/amcl)
    map_server (map_server/map_server)
    move_base (move_base/move_base)
    rviz (rviz/rviz)
    stageros (stage_ros/stageros)

auto-starting new master
process[master]: started with pid [63457]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to de49d692-aaa8-11e7-b290-309c231e60e0
WARNING: Catkin package name "MiniUku_Teb_Planner" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, and underscores.
process[rosout-1]: started with pid [63470]
started core service [/rosout]
process[stageros-2]: started with pid [63494]
process[move_base-3]: started with pid [63495]
process[map_server-4]: started with pid [63496]
process[amcl-5]: started with pid [63497]
process[rviz-6]: started with pid [63506]
[ INFO] [1507302792.237333686]: Loading map from image "/home/iseauto/mini_Uku_ws/src/MiniUku_Teb_Planner/maps/White.png"
[ INFO] [1507302792.238679061]: Read a 200 X 200 map @ 0.050 m/cell
[ INFO] [1507302792.375877775]: Subscribed to map topic.
[ INFO] [1507302792.522438489]: Received a 200 X 200 map @ 0.050 m/pix

[ INFO] [1507302792.530153281]: Initializing likelihood field model; this can take some time on large maps...
[ INFO] [1507302792.540510948]: Done initializing likelihood field model.
[ INFO] [1507302792.945419174, 0.400000000]: Using plugin "obstacle_layer"
[ INFO] [1507302792.970763341, 0.400000000]:     Subscribed to Topics: laser_scan_sensor
[ INFO] [1507302793.850044294, 1.300000000]: Using plugin "obstacle_layer"
[ INFO] [1507302793.871618545, 1.300000000]:     Subscribed to Topics: laser_scan_sensor
[ INFO] [1507302793.961746216, 1.400000000]: Created local_planner teb_local_planner/TebLocalPlannerROS
[ INFO] [1507302794.070914637, 1.500000000]: Footprint model 'line' (line_start: [0,0]m, line_end: [0.7,0]m) loaded for trajectory optimization.
[ INFO] [1507302794.071000970, 1.500000000]: Parallel planning in distinctive topologies enabled.
[ INFO] [1507302794.071032429, 1.500000000]: No costmap conversion plugin specified. All occupied costmap cells are treaten as point obstacles.
[ INFO] [1507302795.364663524, 2.800000000]: Recovery behavior will clear layer obstacles
[ INFO] [1507302795.388241567, 2.800000000]: Recovery behavior will clear layer obstacles
[ INFO] [1507302795.430094569, 2.800000000]: odom received!
[ERROR] [1507303162.893627732, 370.300000000]: NO PATH!
[ERROR] [1507303162.893725523, 370.300000000]: Failed to get a plan from potential when a legal potential was found. This shouldn't happen.
[ERROR] [1507303163.045065321, 370.500000000]: NO PATH!
[ERROR] [1507303163.045403321, 370.500000000]: Failed to get a plan from potential when a legal potential was found. This shouldn't happen.
[ERROR] [1507303163.245820288, 370.700000000]: NO PATH!

croesmann, 2018-06-05:
I think this is more related to the navigation stack in general rather than the teb_local_planner.
The error message is not triggered in this planner.

A side note:
According to your log, I can see that you are using AMCL. AMCL probably will not work properly in empty spaces. You should better provide a static (identity) transformation between `odom` and `map` using a static transform publisher.

closed 2018-06-05 by croesmann

han620, 2020-12-19:
Hi,ghost! I want to use Teb planner without a map ,too.Have you found a solution? @ghost

jcmonteiro, 2020-12-19:
The solution, as Christoph suggested, works. If you don't have a map, your localization provider will (most likely) not work and thus not publish a transform from `map` to `odom`. So, if you don't have a map, you can publish an identity transform yourself.