rst-tu-dortmund/teb_local_planner #60: Influence of the provided Jacobian

beetleskin, 2017-12-05:
I got to know that libg2o supports used-supplied first derivatives, but can also run in derivative-free mode. You already implemented a lot of Jacobians, but [commented them out ](https://github.com/rst-tu-dortmund/teb_local_planner/search?utf8=%E2%9C%93&q=USE_ANALYTIC_JACOBI&type=). In this context, I stumbled upon some questions:

- Why did you comment them out?
- What is the benefit of supplying the Jacobian to the optimizer?
- Could one utilize automatic differentiation (AD, e.g. CppAD or Casadi) for computing them?
- Did you use symbolic tools (matlab, octave, sympy, ..) for calculating them? Do you have sources lying around to continue the work or customize for my own edges?

Thanks, and keep going the impressive work :)

croesmann, 2018-06-04:
Indeed, g2o supports user-supplied first derivatives.
To be honest, I have not experimented much with serious computation time benchmarks related to user-supplied Jacobians.
At an early stage of developing the package, I computed them manually (some by hand and some by symbolic tools (matlab)). However, for some nontrivial cost functions (e.g., penalizing the distance between a line and a polygon) the resulting expressions are super long and for me, it was questionable whether the central difference approximation is slower (2 times evaluating the function) then evaluating the exact Jacobian. But as I sad, I have not performed any benchmarks, so if you have some spare-time, I would be happy to know more about it ;-)

AD is, in general, a great option and it is also very popular in the field of real-time optimal control.
However, to my mind, this would require some major changes in libg2o to provide a high-performance implementation. I played with AD (Casadi) for some optimal control applications and it is indeed suuuper fast as soon as all the preparation is done (the actual graph creation and allocation). In our case, the optimization problem size changes at runtime (insertion and deletion of edges, etc.), so I am not sure about the performance gains here. Do you have any experience?

charon-cheung, 2023-02-07:
@beetleskin Did you try Jacobian or find some other way to improve performance ?