rst-tu-dortmund/teb_local_planner #63: How to set via-points through code?

ben-swain, 2018-01-29:
How can I set via-points through code? The tutorial (http://wiki.ros.org/teb_local_planner/Tutorials/Following%20the%20Global%20Plan%20%28Via-Points%29) only shows how to set them in Rviz.

croesmann, 2018-06-04:
Hi @ben-swain,

There is currently no public method/subscriber to set custom via-points to the local planner navigation plugin.
In case of `test_optim_node` you can send them to the same topic as Rviz is doing.

An implementation would be straightforward since via-points are already supported in the underlying planner code.
I'll see what I can do in the next days, otherwise, I would be happy for any pull request.

croesmann, 2018-06-08:
It is now possible to add via-points via topics! I'll push a new release in a minute ;-)

closed 2018-06-08 by croesmann

marcelpatzsch, 2018-07-27:
I see the changes on the kinetic-devel and also on tag 0.6.10, but when teb_local_planner is installed form the ros repositories at packages.ros.org there is no via_point topic.
When i run `apt list --installed | grep teb-local-planner` i get `ros-kinetic-teb-local-planner/xenial,now 0.6.10-0xenial-20180715-024256-0800 amd64 [installed]` so 0.6.10 is installed and with #74 there should be this option in 0.6.10. Is there any problem i oversee?

croesmann, 2018-08-07:
I tried the same version from the repositories:
```
 ros-kinetic-teb-local-planner/xenial,now 0.6.10-0xenial-20180715-024256-0800 amd64 [installed]
```

The topic `/test_optim_node/via_points` is available in case of running the `test_optim_node`.
For a navigation setup, I invoked `roslaunch teb_local_planner_tutorials robot_diff_drive_in_stage.launch` and got topic `/move_base/TebLocalPlannerROS/via_points`.

Can you still confirm, that you don't have those topics available?

PR #74 (2018-06-08): Adds subscriber for user-supplied via-points
This pull-requests adds the possibility to provide via-points via a topic.
Currently, the user needs to decide whether to receive via-points from the topic or to obtain them from the global reference plan (e.g., activate the latter by setting `global_plan_viapoint_sep>0` as before).

A small test script `publish_viapoints.py` is provided to demonstrate the feature within `test_optim_node`.

This PR fixes #63 .