rst-tu-dortmund/teb_local_planner #66: How to set start and goal?

DanielArnett, 2018-02-28:
I can't find any documentation on how to set the start and goal poses? What topics do I need to set to change them in test_optim_node.launch?

xperroni, 2018-03-02:
As you can see [here](https://github.com/rst-tu-dortmund/teb_local_planner/blob/kinetic-devel/src/test_optim_node.cpp#L167), the start and goal poses are hard-coded, so unless you edit the source code and recompile the node, there's no way to change them.

DanielArnett, 2018-04-28:
Thank you. If anyone wants to just edit this teb_local_planner default app
[image]
[they can change the waypoints here](https://github.com/rst-tu-dortmund/teb_local_planner/blob/kinetic-devel/src/test_optim_node.cpp#L167). 

The better solution for me was to just use the [teb_local_planner_tutorials](https://github.com/rst-tu-dortmund/teb_local_planner_tutorials) and replace the simulated vehicle with my own robot.

Thank  you to everyone who's supported this amazing project.

closed 2018-04-28 by DanielArnett