rst-tu-dortmund/teb_local_planner #67: Error in convertTransRotVelToSteeringAngle() formula?

xperroni, 2018-03-02:
I have seen the conversion formula from steering wheel angle to angular velocity defined as:

    w = v * a / Lf

Where `w` is angular velocity, `v` linear velocity, `a` the steering wheel angle and `Lf` the distance between the rear and from axles. Solving for `a` we get:

    a = Lf * w / v

Which makes perfect sense in terms of units – e.g. if `Lf` is measured in meters, `w` in `rad/s` and `v` in `m/s`, the result will be in `rad`.

However TEB Local Planner's own formula for converting angular velocity to steering angle wheel, as implemented in function `convertTransRotVelToSteeringAngle()`, is:

    a = atan(Lf * w / v)

Which I can't make sense of. Am I missing something?

croesmann, 2018-06-04:
Hi @xperroni,

where does your formula come from? Probably it only works for small angles.
According to the motion of a carlike robot (refer to [1]), the relation is 

    a = atan(Lf / r) 

and `r=v/omega` is the turning radius.

[1] [S. LaValle - Planning Algorithms, Chapter 13.1.2](http://planning.cs.uiuc.edu/)

closed 2018-06-04 by croesmann

xperroni, 2018-06-12:
In fact I find the opposite holds: the version of the formula with `atan` works well for smooth turns that can be made with small steering angles, but it makes impossible to make sharp turns that would require a steering wheel angle close to 90° (or anything much further than 30° really). The reason is that the arc tangent function is asymptotically limited to the interval (-90°, 90°), with a clear slowdown in convergence as early as +/- 45°. To be fair though, I don't know if tricycle robots generally allow the kind of sharp turns my particular application requires, so this may be the more sensible choice after all.

croesmann, 2018-06-13:
to my mind, `a = Lf * w / v`  is still a linearized version that only works for small angles `a`. 
Refer to several papers regarding a (kinematic) non-linear single-track model, bicycle model or carlike model. Their relation is based on a `tan` method. Alternatively, they use the linear verison and directly comment that it is linearized for small angles `a`.

Some sources for tricycle models, derive a model based on the sine-function, e.g. [here](http://users.isr.ist.utl.pt/~mir/cadeiras/robmovel/Kinematics.pdf).

However, you can still implement your own conversion function in a small script/node that subscribes to the twist message published by `move_base`.

xperroni, 2018-06-13:
Thanks for the references, but I must confess I'm not sure what you mean by _"`a = Lf * w / v` is still a linearized version that only works for small angles `a`."_

It's true that trigonometric functions are quasi-linear at small angles, but ultimately what I'm interested at is to compute a steering angle given an angular speed value; therefore my criteria for what "works" is how closely the conversion function tracks this relation.

I made a few tests in simulation, using a tricycle robot with `Lf = 0.96m`. Driving it with linear speed `v = 0.5m/s` and steering angle `a = 1.5` produced an angular speed `w = 0.8rad/s`. Plugging those values to both conversion functions gives:
```
Lf * w / v = 1.536
atan(Lf / r) = 0.993
```
So the linearized version was off by less than 3% while the trigonometric version was off by a third of the actual value. Admittedly this is an extreme case (although not uncommon for my particular application), but it goes to show that contrary to what one might expect, the linearized version does work much better for extreme angles, as well as small ones.

bochen87, 2018-06-13:
If your front steering wheel angle is 1.5 radians (close to 90 degrees), and your robot has a linear speed of 0.5 meters per second, i.e. it is traversing with 0.5 meters per second in the forward (x coordinate in ROS) direction despite the steering wheel being set to almost 90 degrees (blocking towards the left), there are 3 possibilities I can think of:
1) you’re driving very very very fast
2) your vehicle is pushing forward with the back wheels and the front steering wheel is sliding / not making a big difference when put close to a 90 degree angle
3) your formula is wrong

We’ve got tricycle robots running and I can confirm that the formula as is implemented in ten is correct.

croesmann, 2018-06-13:
@xperroni 

> I made a few tests in simulation, using a tricycle robot with ...

It would be interesting to know which tricycle model your simulator actually implements ;-)

I also made a quick test with the stage simulator and their carlike robot model.
I set `Lf = 0.96 m`, `v = 0.5 m/s` and `a = 1.5 rad`.
Stage was not able to actually set the velocity, so the actual measured/simulated velocities are:
`v = 0.035 m/s` and `omega = 0.52 rad/s`.

So it is:

```
Lf * omega / v = 14.26 rad
atan(Lf * omega / v) = 1.50 rad
```

xperroni, 2018-06-13:
The 0.5m/s figure is just a value I picked for a quick simulation test. We do however turn at 1.5 rad and 0.25m/s with a real tricycle robot, and I can assure you there is no sliding.

I don't dispute the trigonometric formula works well for in the general cases, all I'm pointing out is that it breaks down for extreme angles, and the linearized formula works much better for such cases.

Also to be clear, I'm not advocating the TEB implementation be changed, I'm just reporting a use case for reference purposes. We used TEB in the past but have since moved to our own custom local planner, so we're not really concerned with this anymore.

bochen87, 2018-06-13:
I am assuming that you have a tricycle with 2 back wheels and one front wheel.

Is v in your definition the linear velocity of your robot or is v the speed of you front wheel? That makes a big difference.

I can hardly imagine a robot going with 0.25 m/s linear velocity when the front wheel has an 86 degree steering angle set.