rst-tu-dortmund/teb_local_planner #70: can't use bestTeb() to access poses of computed local path

GuillaumeHauss, 2018-03-26:
Hi all, 

I would like to add a feature to the planner: publishing only the local path, without any velocity commands. That will be done by an external node.
However, when I write
```cpp
TebOptimalPlannerPtr selected_teb = planner_->bestTeb();
```
I got the error 
```cpp
error: ‘class teb_local_planner::PlannerInterface’ has no member named ‘bestTeb’
   TebOptimalPlannerPtr selected_teb = planner_->bestTeb(); 
```

which I know is false as it's clearly defined in homotopy_class_planner.h (l.208)

Any idea why? 

Thank you very much for your help

GuillaumeHauss, 2018-03-27:
Okay, it seems that there's a lot of things to do to add a new function to a local planner..
I'm adding declarations here and there but I think I'm just messing things up

Can one of you guys guide me through the required steps to add a new function to the local planner? 

Thank you very much

croesmann, 2018-06-01:
Hi!
I suggest to first investigate the complete code of the package.
The actual planner is defined in `optimal_planner.cpp` and it inherits from the abstract `PlannerInterface` class.
The `HomotopyClassPlanner` is just an extension that maintains multiple planners.

If you set parameter `enable_homotopy_class_planning` to true, the local planner plugin instantiates a HomotopyClassPlanner. Otherwise, `TebOptimalPlanner` is instantiated directly.
If you want to access your method at the plugin level, you might also extend the PlannerInterface.

However, the local plan is already being published by the navigation stack?

Cheers,
Christoph

GuillaumeHauss, 2018-06-04:
Hi Christoph, 
Thank you for answering me

It's true that the local plan is already published, as a feedback/visualization tool. 
My idea was to create a new function that will only compute the path and not the velocity commands, as they are jumpy and not really precise.
However, after some clocking, it seems that the path computing -- which is a recursive process -- takes  nearly 90% of the time. So getting rid of the lasting 10% doesn't appear that necessary now.

Maybe I'll come back to it when I'll have some more free time, work is crazy right now 

Cheers
Guillaume