rst-tu-dortmund/teb_local_planner #71: On the Spot Turing Problem in Differential Drive Robot

dharmp15, 2018-05-02:
Hi,
After update of #54 on the spot turning is not happening in Diff drive robot. In older versions it was fine. It seems the min_turning_radius=0 is not effective and planner is treating robot as car like only. Did anyone face this problem after using latest kinetic version?? Any suggestion most welcome.

croesmann, 2018-06-01:
Hi,
Thanks for reporting the issue! 
I tried to reproduce this issue. The planner correctly calls the diff-drive methods (in particular it adds diff-drive cost edges). However, if I compare both versions with the same parameter settings, I can also observe slightly different planning behaviors. This is indeed not intended. I will try to find the reason for this.

croesmann, 2018-06-04:
I think I found the issue and it goes back to commit 270d1daeaef069fc204978112e1ff08d7f698bb5.
It should be fixed now and if not, please reopen this issue.

closed 2018-06-04 by croesmann

dharmp15, 2018-06-06:
Thanks for your effort but it seems problem is not resolved. I checked with your modifications.

croesmann, 2018-06-06:
I tried it with `roslaunch teb_local_planner_tutorials robot_diff_drive_in_stage.launch`.
It works pretty well. Please note, turning on the spot is **not** the optimal solution for the default parameter setting, since the maximum allowed rotational velocity is low.
Increase `weight_kinematics_forward_drive` and `max_vel_theta` if you would like to prefer turns on the spot.

dharmp15, 2018-06-13:
Thanks. increased weight_kinematics_forward_drive parameter and now working  fine.