rst-tu-dortmund/teb_local_planner #78: Parameters for making a car-like vehicle only plan movement backwards

lucascoelhof, 2018-06-20:
Hello folks,

First of all, thanks for making this great package available! So far it's been really useful for me.

I'm running this package in a autonomous vehicle simulator, on ROS Indigo. In a specific use-case I have, I want the car-like robot to plan a reverse maneuver to a parking spot. The distance from the vehicle and the parking spot can be even a few hundred meters, but nevertheless, in this use-case it should only move backwards, even if it would be easier to go forward. Are there a combination of parameters that would allow this? I've been tweaking with parameters "weight_kinematics_forward_drive", "max_vel_x_backwards" and "wheelbase", but I was unable to make the robot only drive backwards.

croesmann, 2018-06-20:
Hi Lucas,

thanks a lot for your feedback.
I think the question could be interesting for more users, so I have added a new section to the [FAQ (Following global plan backwards)](http://wiki.ros.org/teb_local_planner/Tutorials/Frequently%20Asked%20Questions#Following_global_plan_backwards) in the tutorial section.
So in summary, it is a matter of the global planner to my mind. 

However, I would be happy to discuss, if it would make sense to add further strategies to the local planner (assuming `global_plan_overwrite_orientation=true`), which apply to a wider range of practical applications. 

Cheers,
Christoph

lucascoelhof, 2018-06-27:
Hi @croesmann , I got the chance to test the suggested parameters, it worked as I wanted, thank you for your attention! What I'm doing now is whenever I need to do that reverse maneuver, I call dynamic reconfigure to set the parameters for planning on reverse and on forward movement I restore the old ones.

closed 2018-06-27 by lucascoelhof