rst-tu-dortmund/teb_local_planner #82: Straight global plans do not obey 'global_plan_viapoint_sep' nor 'max_global_plan_lookahead_dist'

varandaas, 2018-06-29:
Whenever I send a straight global plan, TEB doesn't comply with these parameters and generates the whole trajectory until the goal. 

Good behavior with a curved trajectory:
[image]

Bad behavior with a straight trajectory:
[image]

Is there any parameter to solve this?

If not, how easy would it be to fix it and create a parameter to solve this?

croesmann, 2018-06-29:
Hi, 
that definitely looks like a bug. Can you please test my code from branch `global_plan_mod`. I did some changes their which might have an effect here.
Otherwise, I will check the code again.

Cheers,
Christoph

varandaas, 2018-06-29:
Hello Christoph,

Thank you so much for your help! I've tested it on a couple examples and that branch solves the distance issue, given a straight trajectory. However, I noticed two behaviors that don't seem correct:

- It still doesn't respect the local costmap boundaries (screenshot below);
- I was also expecting it to follow that trajectory with viapoints like curved trajectories (and respecting the  'global_plan_viapoint_sep'), but couldn't make it.

[image]

Are you planning to merge this fix to the following releases? 

Can I help you further with testing?

Best,
Francisco

croesmann, 2018-06-29:
I probably have another reason: how many points define your straight line? If it is just two, the behavior is reasonable. The local planner just iterates all points from the local plan until it is outside of the local costmap or the max dist is exceeded (this is similar to other local planners). In case of just two points the search will stop at the goal. 

You need to interpolate your straight line on a fine grained grid

croesmann, 2018-06-29:
Yes I will merge that code and check your recent issues early next week

varandaas, 2018-06-29:
You're right @croesmann, my straight line is 150 meters and my interpolation had only 20 points (7,5 meters between pts vs 8x8 meter local costmap and 3 meter 'max_global_plan_lookahead_dist'). Your explanation makes total sense, my mistake.

Thank you so much for the fixes :) Looking forward to the release.

PR #84 (2018-07-05): Slightly changing the behavior of how the global plan and via-points are processed
Global plan:
In every sampling interval, the global plan is processed in order to find the closest pose to the robot (as reference start) and the current end pose (either local at costmap boundary or `max_global_plan_lookahead_dist`). Previously, the search algorithm stopped as soon as the distance to the robot increased once. This caused troubles with more complex global plans, hence the new strategy checks the complete subset of the global plan in the local costmap for the closest distance to the robot.

This  [... cut]
closed 2018-07-05 by croesmann