rst-tu-dortmund/teb_local_planner #83: ROS Melodic

AchmadFathoni, 2018-07-01:
Hi,  since there is still no official repository for melodic what is alternative for properly install teb_local_planner on melodic?

croesmann, 2018-07-01:
Hi,
the navigation stack has not been released for melodic yet (see [here](https://github.com/ros-planning/navigation/issues/749) and hence is still [blocking](http://repositories.ros.org/status_page/blocked_releases_melodic.html?q=teb&s=6&r=1) the teb_local_planner package. As soon as it is released, I'll release the teb_local_planner and costmap_converter.

Cheers,
Christoph

boilerbots, 2018-07-23:
The tf2 changes to the costmap_2d being made for melodic need to be implemented, code no longer compiles.

mintar, 2018-08-07:
Navigation has been released into melodic: https://github.com/ros/rosdistro/pull/18657

So there's nothing blocking a teb_local_planner release into Melodic now.

AchmadFathoni, 2018-08-08:
@croesmann Thank you very much!

closed 2018-08-08 by AchmadFathoni