rst-tu-dortmund/teb_local_planner #85: Forbid moving backwards completely?

DingFeng0103, 2018-07-10:
Hi,

I am a big fan of your work. I have tuned `weight_kinematics_forward_drive` to maximum e.g. 1000. But is still sometimes move backwards. Is there a way to completely ban this behavior?

Thanks
Feng

DingFeng0103, 2018-07-10:
Hi,
As i was checking your code, I found `penalty_epsilon` this particular parameter has no use, I mean I couldn't found a  .cpp file that utilize it. Could you tell me what this parameter is doing? and this parameter limit my `max_vel_x_backwards` so that I have to set `max_vel_x_backwards` to be bigger than `penalty_epsilon`

DingFeng0103, 2018-07-11:
Since  I found `penalty_epsilon` has no use. I set it to a very small value like 0.00001 so that I can make `max_vel_x_backwards`  really small thus backwards doesn't satisfy time optimal condition. Problem seemed solved, though I wonder if this is the right way to do it.

varandaas, 2018-07-31:
I'm also interested in this issue. I tried your solution @DingFeng0103 , but it still tried to go backward. Can you share your parameters?

DingFeng0103, 2018-08-01:
sure, I can't promise backwards move is completely forbidden, but now it hardly happens.
```

TebLocalPlannerROS:

 odom_topic: odom
 map_frame: /odom
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5
    
 # Robot
         
 max_vel_x: 0.25
 max_vel_x_backwards: 0.01
 max_vel_theta: 0.4
 acc_lim_x: 0.06
 acc_lim_theta: 0.04
 min_turning_radius: 0.0
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
   type: "line"
#   radius: 0.2 # for type "circular"
   line_start: [-0.1, 0.0] # for type "line"
   line_end: [0.1, 0.0] # for type "line"
#   front_offset: 0.2 # for type "two_circles"
#   front_radius: 0.2 # for type "two_circles"
#   rear_offset: 0.2 # for type "two_circles"
#   rear_radius: 0.2 # for type "two_circles"
#   vertices: [ [0.25, -0.05], [0.18, -0.05], [0.18, -0.18], [-0.19, -0.18], [-0.25, 0], [-0.19, 0.18], [0.18, 0.18], [0.18, 0.05], [0.25, 0.05] ] # for type "polygon"

 # GoalTolerance
    
 xy_goal_tolerance: 0.05
 yaw_goal_tolerance: 0.05
 free_goal_vel: False
    
 # Obstacles
    
 min_obstacle_dist: 0.3
 inflation_dist: 1.0
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: False
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 4
 no_outer_iterations: 3
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.001
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1000
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 100
 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet
 alternative_time_cost: False # not in use yet

 # Homotopy Class Planner

 enable_homotopy_class_planning: False
 enable_multithreading: False
 simple_exploration: False
 max_number_classes: 4
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False
```

varandaas, 2018-08-03:
Thank you @DingFeng0103, I'll test it :+1:

croesmann, 2018-08-07:
This is currently the correct way, as I indicated [here](https://answers.ros.org/question/272462/how-to-prevent-navigate-backwards-and-only-rotate-and-move-forwards-for-navigation/) as well.
Unfortunately, we cannot enforce it hard since our optimizer only handles soft constraints and hence finds a tradeoff between all weighted objectives and penalties. Too large weights would imply a bad condition of the optimization problem and hence worse convergence speeds.

For non car-like (diffdrive and omni) robots, we could overwrite the outcome of the optimzier by just performing an in-place rotation until the solution with the backwards motion transforms into a solution with only a forward motions (implementation should be straight-forward). If this could help someone, I am happy for any pull request ;-)

varandaas, 2018-08-07:
Thanks, @croesmann, this would help me a lot and I'm quite interested in contributing to it for non-car-like robots. Do you have some more hints, so I can speed up the development?

croesmann, 2018-08-07:
Actually, I already started a few minutes ago looking at it ;-). I have 1-2 hours spare-time today, so will see what I can accomplish ;-). I'll give you an update here later, and if it is not finished by then, feel free to take over.

croesmann, 2018-08-09:
I started implementing this in branch `enforced_in_place_rotation`. 
It works in many situations, however, not in all, because it can cause oscillations in conjunction with the optimizer (e.g. by placing the goal behind the robot). The question is, whether we are able to find a complete solution without lots of special cases to check (I actually don't like to mess things up with tons of rules ;-) the best solution would still be to use hard-constraints, which is currently not possible with the underlying solver).
I currently have not enough spare-time to continue to work on this. So please feel free to take over if you like.

varandaas, 2018-08-09:
Thank you so much for your effort @croesmann :D I totally agree with you, we should find a smooth solution without those special cases. I will try to understand it and take over, but it will probably take some weeks. Thanks!

amakarow, 2020-10-17:
Closed because #230 was merged and allows larger weights.

closed 2020-10-17 by amakarow