rst-tu-dortmund/teb_local_planner #91: Navigation problems with UGV

Anannf, 2018-08-16:
Hi,
My team has been using the teb local planner in our UGV program, and we have encountered a couple problems.

Firstly, the car has demonstrated overshooting of steering. On long and smooth pathes, the car would attempt to turn and couldnot follow the plan smoothly. It seems that when we set `cmd_angle_instead_rotvel` to false, this problem wouldn't occur.

Secondly, the vehical tries to move back and forth constantly on straight pathes with no obstacles. The path was just a smooth line that is almost straight, but the car just cannot follow it.

I havn't found any solutions in the forums or solved Github Issues, can anyone help?

Our parameters are listed below.

>acc_lim_x: 0.5
acc_lim_theta: 0.05
max_vel_x: 0.7
max_vel_x_backwards: 0.23
max_vel_theta: 0.7

>#carlike robots
min_turning_radius : 3
wheelbase: 2.49
cmd_angle_instead_rotvel: true

>footprint_model:
  type: "polygon"
  vertices: [ [2, -0.7], [2, 0.7], [-0.45, 0.7], [--0.45, -0.7] ]
  

>xy_goal_tolerance: 1.5
yaw_goal_tolerance: 3.0
free_goal_vel: false

>#trajectory
dt_ref: 0.3
dt_hysteresis: 0.1
#obstacle
min_obstacle_dist: 0.5
include_costmap_obstacles: false
inflation_dist: 0.6

>#optime:
no_inner_iterations: 5
no_outer_iterations: 4
penalty_epsilon: 0.01
weight_max_vel_x: 2.0
weight_max_vel_theta: 1.0
weight_acc_lim_x: 1.0
weight_acc_lim_theta: 5.0
weight_kinematics_nh: 1000.0
weight_kinematics_forward_drive: 1.0
weight_kinematics_turning_radius: 1.0
weight_optimaltime: 1.0
weight_obstacle: 200.0
weight_viapoint: 1.0
weight_inflation: 0.5
weight_adapt_factor: 2.0

>publish_feedback: true
enable_homotopy_class_planning: true
odom_topic: /golfcar/odom
map_frame: world
include_dynamic_obstacles: false
force_reinit_new_goal_dist: 0.1
reinit_time: -1
obstacle_heading_threshold: -0.3
allow_init_with_backwards_motion: true
roadmap_graph_no_samples: 3
roadmap_graph_area_width: 8
selection_cost_hysteresis: 1.1

subodh-malgonde, 2018-10-26:
I am facing a similar problem - oscillations as well as overshoot. I posted a question on ROS Answers [Oscillations when using teb_local_planner for car like robot ](https://answers.ros.org/question/306859/oscillations-when-using-teb_local_planner-for-car-like-robot/).

@Anannf  Were you able to fix this?

Anannf, 2018-10-28:
Hi @subodh-malgonde. I was just an intern at the project and I have left now. However, I can tell you that my former colleagues have successfully fixed the problem by rewriting the teb planner, purely based on the principle proposed in the paper.

closed 2018-10-28 by Anannf

subodh-malgonde, 2018-11-01:
@Anannf You were right. I should have guessed that something is limiting the steering, hence the back and forth. The following changes fixed it for me:

1. Decreased inflation_dist and min_obstacle_dist
2. Increased max_vel_theta and acc_lim_theta