rst-tu-dortmund/teb_local_planner #95: Obstacle avoidance problem

joehrobot, 2018-08-23:
Hi,
here is my illustration:
When robot do obstacle avoidance plan, sometime teb local planner plan a weird rotation.
like following picture, blue one is robot , black circle is a weird rotation trajectory.
[image]

this my setting for teb local planner:

TebLocalPlannerROS:

 odom_topic: odom

 # Trajectory

 teb_autosize: True
 dt_ref: 0.5
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 global_plan_viapoint_sep: 0.6

 allow_init_with_backwards_motion: False
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5

 # Robot

 max_vel_x: 0.38
 max_vel_x_backwards: 0.2
 max_vel_y: 0.0
 max_vel_theta: 0.5
 acc_lim_x: 0.2   # define the max acc
 acc_lim_theta: 0.25   #  define how fast reach max_vel_x# 1.5
 min_turning_radius: 0.0 # diff-drive robot (can turn on place!)

 #footprint_model:
   #type: "point"
 #is_footprint_dynamic: True
 # GoalTolerance

 xy_goal_tolerance: 0.07
 yaw_goal_tolerance: 0.03
 free_goal_vel: False

 # Obstacles

 min_obstacle_dist: 0.35 #  for Jackal This value must also include our robot radius, since footprint_model is set to "point".
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.5
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5
 # Optimization

 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1000
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_viapoint: 10.0

 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet
 weight_adapt_factor: 2

 # Homotopy Class Planner

 enable_homotopy_class_planning: False
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 4
 selection_cost_hysteresis: 1.0
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False

 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False

thanks for ur help!

croesmann, 2018-10-30:
thanks for reporting this. To my mind, the obstacle pushed the trajectory in such an "undesired" way, that it winds up. The optimization result seems to a local minimum. In my cases, the optimizer was often able to slowly recover due to the time-optimality cost (the winded trajectory shrinks to a single point, i.e., an in-place rotation, which can then easy resolved). However, in this case, it seems that the optimizer keeps the winded trajectory. 

Besides some careful parameter tuning (which might not be an ultimate solution), one could detect winded trajectories and replace them by a new initial trajectory (warm-start) that does not contain any winded parts anymore. However, this is currently not in the code base. Any merge-request is welcome ;-)

amakarow, 2020-10-17:
I closed it for now as #139 tries to fix it. Otherwise feel free to comment.

closed 2020-10-17 by amakarow