rst-tu-dortmund/teb_local_planner #100: Teb planner didn't turn at right angle which lead to y_axis deviation from goal point

DingFeng0103, 2018-10-11:
Not sure if anyone else met the same problem. When I use teb for diff-robot navigation the robot should make a turn then reach the goal point however it didn't turn much enough thus leave a y-axis deviation from goal point then it start to go back and forth in order to reduce the y-axis deviation which is hard for diff-robot. when I use teb for car-like robot this problem is more obvious since it is even harder for car-like robot to reduce y-axis deviation.

DingFeng0103, 2018-10-11:
Another question is that when I check the code, I found forward_driving_weight is not applied to car-like robot in `AddEdgesKinematicsCarlike`  function. So how can I force the robot to drive forward as much as possible for car-like robot?

croesmann, 2018-10-30:
for the first question: this seems to be a model mismatch between the planner kinematic model and your real robot (maybe friction, or wrong parameters for velocity and acceleration limits).

for the second question: I left this out since a strict penalty (soft-constraint) can make the condition of the optimization problem worse (sometimes, the car really needs to go backward!). 

However, I agree, adding a small penalty might be reasonable to prefer forward driving. But the weight should be low. Please feel free to add the penalty and test the behavior. I would be happy to merge any pull request if this works as expected.