rst-tu-dortmund/teb_local_planner #103: Discrepancy between the initial poses of teb local plan and global plan

fwthree3, 2018-11-05:
Hi, 

I have an issue in using TEB planner, which is about a discrepancy of starting poses between global and local path. The local plan from TEB planner in my omni-directional mobile robot doesn't start from either the current pose or the start position of the global path, especially when time goes by. The phenomenon does not appear at initial but it is shown up later on. 

Let me attach a gif image to describe the problem. 

[image]

In the gif, the robot's pose (with its covariance) is shown in a red arrow, global plan is the green line, local plan is the blue line, and the teb poses for future time steps are the yellow arrows. The actual robot is placed where the pose in gif is shown, and the odom topic is also shown in the same place. 

The gif shows a situation when I moved the robot to the sideways (this is an omni-directional robot). You will see the discrepancy of the global(green) and local path(blue) at their starting poses in the gif. For example, if I move the robot to the right, the wrong pose (one in the yellow arrows) of the starting point of the local path (blue) moves the right. The final poses seem to be the same, so I am thinking this happening is due to the wrong selection for the starting pose of the local planner. I tried a different planner (DWA) and the phenomenon didn't occur. 

I wonder if you have seen or been reported about this issue. It will be also very appreciative if you let me anything about possible reasons of this phenomenon. I might be wrong for using the planner such as wrong setting, but at this moment I think this is not the setting problem. 

Thank you very much. 

Jw Han

croesmann, 2018-11-07:
Hi @fwthree3,
this looks like some TF discrepancy. Can you tell me which frames you are using in your navigation setup, especially your local costmap settings.
Can you also compare your settings with my example package [teb_local_planner_tutorials](https://github.com/rst-tu-dortmund/teb_local_planner_tutorials). There is a configuration for an omni-directional robot.

Afterward, we can try to figure out if this is just your configuration or if I have some wrong or missing tf call in [teb_local_planner_ros.cpp](https://github.com/rst-tu-dortmund/teb_local_planner/blob/kinetic-devel/src/teb_local_planner_ros.cpp).

fwthree3, 2018-11-07:
Hi, 

I checked all parameters related to local costmap and TEB planner by using "rosparam get" command while implementing omni_tutorial and my own one. 
 
For local costmap, below is the related frames in my configuration. 
      - /move_base/local_costmap/global_frame : odom
      - /move_base/local_costmap/robot_base_frame: base_footprint
I found that "local_costmap/global_frame" in TEB tutorial is set as "map", and tried to change my one from "odom" to "map". It doesn't resolve the phenomenon though. 

For TEB omni setup, I compared each parameter between the tutorial and my own one. Below is the difference. 
1) For "footprint_model", I set the "type" as "line" with the information for the line. I use a plugin, CostmapToPolygonsDBSMCCH, for a costmap converter.
2) I set "map_frame" as "map"
3) I set "odom_topic" as "odom"

I attach a capture showing "tf" for "base_footprint" and "odom" for my system (please note that they are overlapped in position). Please ignore the discrepancy between the costmap and map shown in the capture. It is because I arbitrary reset the initial pose to reenact the phenomenon of discrepancy of the paths. (green line: global path, blue line: local path, red arrows: teb poses)

[image]

Thank you very much,

Jw Han

croesmann, 2019-04-10:
sorry for my very late response. Is this still relevant? If so, let me know and we can reopen the issue

closed 2019-04-10 by croesmann