rst-tu-dortmund/teb_local_planner #107: Robot gets stuck entering a narrow pathway

subodh-malgonde, 2018-11-13:
I have a car-like robot. It navigates very well in areas where obstacles are farther away. However it sometimes gets stuck in narrow spaces and has a problem entering narrow pathways. See the below video to visualize the problem in Rviz. 

**Click on the image to play the video. Redirects to YouTube**
[image]

I get the error:

> TebLocalPlannerROS: trajectory is not feasible. Resetting planner...

These are my parameters:

```
TebLocalPlannerROS:

 # Odometry information that gives the local planner the current speed of the robot.
 odom_topic: /vesc/odom

 # Global planning frame . Default value is "odom".
 # (in case of a static map, this parameter must be usually changed to "/map".
 map_topic: /map

 ########## Trajectory ##########
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 allow_init_with_backwards_motion: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 4
    
 ########## Robot ##########
         
 max_vel_x: 1.5
 max_vel_x_backwards: 0.7
 max_vel_y: 0.0
 max_vel_theta: 2.5 # the angular velocity is also bounded by min_turning_radius in case of a carlike robot (r = v / omega)
 acc_lim_x: 6.0
 acc_lim_theta: 4.0

 # ********************** Carlike robot parameters ********************
 min_turning_radius: 1.25        # Min turning radius of the carlike robot (compute value using a model or adjust with rqt_reconfigure manually)
 wheelbase: 0.34                 # Wheelbase of our robot
 cmd_angle_instead_rotvel: False # stage simulator takes the angle instead of the rotvel as input (twist message)
 # ********************************************************************

 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
   type: "polygon"
   vertices: [ [-0.28,-0.15], [0.28,-0.15], [0.28,0.15], [-0.28, 0.15] ]

 ########## GoalTolerance ##########
    
 xy_goal_tolerance: 0.2
 yaw_goal_tolerance: 0.1

 # Remove the goal velocity constraint such that the robot can arrive at the goal with maximum speed
 free_goal_vel: False
    
 ########## Obstacles ##########
    
 min_obstacle_dist: 0.15 # This value must also include our robot's expansion, since footprint_model is set to "line".
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0

 # Buffer zone around obstacles with non-zero penalty costs (should be larger than
 # min_obstacle_dist in order to take effect). Default is 0.6.
 inflation_dist: 0.10
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 ########## Optimization ##########
    
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 10.0
 weight_max_vel_theta: 10.0
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 10
 weight_kinematics_turning_radius: 50
 weight_optimaltime: 1
 weight_obstacle: 50

 ########## Homotopy Class Planner ##########

 enable_homotopy_class_planning: False
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 2
 selection_cost_hysteresis: 1.0
 selection_obst_cost_scale: 1.0
 selection_alternative_time_cost: False
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False

```

costmap_common_params.yaml

```
footprint: [ [-0.28,-0.15], [0.28,-0.15], [0.28,0.15], [-0.28, 0.15] ]

transform_tolerance: 0.2
map_type: costmap

obstacle_layer:
 enabled: true
 obstacle_range: 3.0
 raytrace_range: 3.5
 inflation_radius: 0.10
 track_unknown_space: false
 combination_method: 1

 observation_sources: laser_scan_sensor
 laser_scan_sensor: {data_type: LaserScan, topic: scan, marking: true, clearing: true}

inflation_layer:
  enabled:              true
  cost_scaling_factor:  10.0  # exponential rate at which the obstacle cost drops off (default: 10)
  inflation_radius:     0.10  # max. distance from an obstacle at which costs are incurred for planning paths.

static_layer:
  enabled:              true
  map_topic:            "/map"
```

Been struggling with this problem for more than 2 weeks now. Tried reducing `inflation_radius` and `min_obstacle_dist`, did not help my case.

stevemartinov, 2019-03-06:
How did you tune the angular rotation of front axes wheels?

croesmann, 2019-04-10:
Your obstacle avoidance settings are not configured properly as indicated [here](http://wiki.ros.org/teb_local_planner/Tutorials/Frequently%20Asked%20Questions#Gaps_in_the_trajectory).
There is also a good answer below your video ;-)

closed 2019-04-10 by croesmann