rst-tu-dortmund/teb_local_planner #114: move_base crash when using teb_local_planner

rhklite, 2019-01-08:
I followed ROS tutorial on adding teb_local_planner to my move_base. However move_base crashes.

**DISPLAYED ERROR**
```bash
[ INFO] [1546979791.782741452]: Created local_planner teb_local_planner/TebLocalPlannerROS
[ WARN] [1546979791.894919713]: TebLocalPlannerROS() Param Warning: 'alternative_time_cost' is deprecated. It has been replaced by 'selection_alternative_time_cost'.
[ INFO] [1546979791.906846064]: Footprint model 'circular' (radius: 0.2m) loaded for trajectory optimization.
[ INFO] [1546979791.907151616]: Parallel planning in distinctive topologies disabled.
[ INFO] [1546979791.923209105]: Costmap conversion plugin costmap_converter::CostmapToPolygonsDBSMCCH loaded.
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
[move_base-23] process has died [pid 10433, exit code -6, cmd /home/usar/catkin_ws/devel/lib/move_base/move_base odom:=/zed/odom_relay scan:=/scan __name:=move_base __log:=/home/usar/.ros/log/0f3b9c0e-1385-11e9-b099-b8aeed739cc6/move_base-23.log].
log file: /home/usar/.ros/log/0f3b9c0e-1385-11e9-b099-b8aeed739cc6/move_base-23*.log
```

**move_base launch file**
```xml
<?xml version="1.0"?>
<!--
    ROS navigation stack with velocity smoother and safety (reactive) controller
-->
<launch>

  <arg name="odom_frame_id"   default="zed_odom"/>
  <arg name="base_frame_id"   default="base_link"/>
  <arg name="global_frame_id" default="map"/>
  <arg name="odom_topic" default="/zed/odom_relay" />
  <arg name="laser_topic" default="/scan" />
  <arg name="custom_param_file" default="$(find jaguar_navigation)/param/teb_local_planner.yaml"/>

  <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <rosparam file="$(find jaguar_navigation)/param/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find jaguar_navigation)/param/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find jaguar_navigation)/param/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find jaguar_navigation)/param/global_costmap_params.yaml" command="load" />
    <!--rosparam file="$(find jaguar_navigation)/param/dwa_local_planner_params.yaml" command="load" /-->
    <rosparam file="$(find jaguar_navigation)/param/move_base_params.yaml" command="load" />
    <rosparam file="$(find jaguar_navigation)/param/global_planner_params.yaml" command="load" />
    <rosparam file="$(find jaguar_navigation)/param/navfn_global_planner_params.yaml" command="load" />
    <!-- external params file that could be loaded into the move_base namespace -->
    <rosparam file="$(arg custom_param_file)" command="load" />

    <param name="base_local_planner" value="teb_local_planner/TebLocalPlannerROS" />
    <param name="controller_frequency" value="5.0" />

    <!-- reset frame_id parameters using user input data -->
    <param name="global_costmap/global_frame" value="$(arg global_frame_id)"/>
    <param name="global_costmap/robot_base_frame" value="$(arg base_frame_id)"/>
    <param name="local_costmap/global_frame" value="$(arg odom_frame_id)"/>
    <param name="local_costmap/robot_base_frame" value="$(arg base_frame_id)"/>
    <param name="DWAPlannerROS/global_frame_id" value="$(arg odom_frame_id)"/>

    <!-- Correcting angular velocity through velocity_transform node before sending to jaguar -->
    <!-- <remap from="cmd_vel" to="/jaguar_velocity_controller/cmd_vel"/> -->
    <!--remap from="/map" to="/rtabmap/grid_map" /-->

    <remap from="odom" to="$(arg odom_topic)"/>
    <remap from="scan" to="$(arg laser_topic)"/>
  </node>
</launch>
```

```yaml
TebLocalPlannerROS:

 odom_topic: /zed/odom_relay
 map_frame: /map

 # Trajectory

 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5

 # Robot

 max_vel_x: 0.4
 max_vel_x_backwards: 0.2
 max_vel_theta: 0.3
 acc_lim_x: 0.5
 acc_lim_theta: 0.5
 min_turning_radius: 0.0
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
   type: "circular"
   radius: 0.2 # for type "circular"
   line_start: [-0.3, 0.0] # for type "line"
   line_end: [0.3, 0.0] # for type "line"
   front_offset: 0.2 # for type "two_circles"
   front_radius: 0.2 # for type "two_circles"
   rear_offset: 0.2 # for type "two_circles"
   rear_radius: 0.2 # for type "two_circles"
   vertices: [ [0.25, -0.05], [0.18, -0.05], [0.18, -0.18], [-0.19, -0.18], [-0.25, 0], [-0.19, 0.18], [0.18, 0.18], [0.18, 0.05], [0.25, 0.05] ] # for type "polygon"

 # GoalTolerance

 xy_goal_tolerance: 0.2
 yaw_goal_tolerance: 0.1
 free_goal_vel: False

 # Obstacles

 min_obstacle_dist: 0.4
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization

 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
```

croesmann, 2019-04-10:
The behavior is somewhat strange.
According to your log you are loading the costmap converter plugin:
```
[ INFO] [1546979791.923209105]: Costmap conversion plugin costmap_converter::CostmapToPolygonsDBSMCCH loaded.
terminate called after throwing an instance of 'std::bad_alloc'
```
However, the config file you posted contains `costmap_converter_plugin: ""`.
Are you sure that you are loading the correct config file?

richban, 2020-07-29:
I have a similar [issue](https://github.com/rst-tu-dortmund/teb_local_planner_tutorials/issues/13) however I am not loading any conversion plugin.

remember505, 2021-05-29:
I also encountered this problem. How can I solve it？ Do you now why? @croesmann @richban @rhklite

remember505, 2021-05-29:
> I have a similar [issue](https://github.com/rst-tu-dortmund/teb_local_planner_tutorials/issues/13) however I am not loading any conversion plugin.

Hello，is this problem solved?

richban, 2021-05-31:
Can't recall but I believe the issue was regarding wrong version or distribution installed

thugofROS, 2021-07-08:
same problem could some one help me solve it .

tomy983, 2022-07-10:
same for me...

tomy983, 2022-07-10:
Here is how I fixed it for noetic:
```
sudo apt remove ros-noetic-teb-local-planner
cd catkin_ws/src
git clone -b noetic-devel https://github.com/rst-tu-dortmund/teb_local_planner.git
cd ..
catkin_make
rosdep install teb_local_planner
```

rohit4myself, 2022-08-26:
> Here is how I fixed it for noetic:
> 
> ```
> sudo apt remove ros-noetic-teb-local-planner
> cd catkin_ws/src
> git clone -b noetic-devel https://github.com/rst-tu-dortmund/teb_local_planner.git
> cd ..
> catkin_make
> rosdep install teb_local_planner
> ```

wasted 4 hours resetting my workspace, testing with previous commits..You comment saved another 4. Thanks.