rst-tu-dortmund/teb_local_planner #117: It seems that PolygonObstacle does not work?

peterWon, 2019-02-25:
Hi, sorry to disturb and thanks a lot for your great work.
I try to add obstacle in data type of 'PolygonObstacle', but the planned path still goes through the obstacle, but when I change the type to  'PointObstacle', the planner works well, I have no idea where goes wrong.
The type 'PointObstacle' can't exactly descrobe the obstacle in my situation. Would you please give me some tips, thanks a lot!

croesmann, 2019-04-10:
If this is still relevant, can you provide me with a screenshot?
Can you test if the following scenario works because it includes a PolygonObstacle (you need to run each command in a separate terminal):
```
roslaunch teb_local_planner test_optim_node.launch
rosrun teb_local_planner publish_test_obstacles.py 
```

peterWon, 2019-05-22:
@croesmann Hi, croesmann, Here are two situation which teb does not works well(I used PointObstacle to describe the obstacles).
[image]
[image]