rst-tu-dortmund/teb_local_planner #118: Not able to specify via points

msadowski, 2019-03-02:
I've been following the tutorial for setting via_points through RVIZ.

Working on the first part of the tutorial and using _roslaunch teb_local_planner test_optim_node.launch_ the path is updated as expected, however when following the tutorial further and executing _roslaunch teb_local_planner_tutorials robot_diff_drive_in_stage.launch_ the points set through Publish Point in rviz are not showing up (I've tried setting the global_plan_viapoint_sep to both -0.1 and 0.5).

I tried placing the points just after rviz started and also while the robot is navigating to a 2D pose but I don't see any points show up in RVIZ (listening to /clicked_point I can see there are correctly sent).

Is there anything that I might be missing? I'm happy to make PR to the tutorial docs if there is any missing step there.

I'm currently on Kinetic and I've tested it both with teb_local_planner installed via apt-get and compiled from source.

croesmann, 2019-04-10:
Hi, sorry for my late response.
The implementation of providing via points through code is not completely finished
as the most common use case for me was to obtain them from the global plan online.
Could you check if the information in #102 , #63 and #74 already solves your problems? 
Otherwise, we could extend interfaces to improve via point support.

r91andersson, 2019-09-17:
We're running a setup where we have recorded the robots path over time, and then let it be played back to the robot, but as goals. When we are close to one of the goals, we send the next goal (i.e interrupt the path planning) to make it follow the recorded path. However, doing this makes the teb_local_planner stop sending out the cmd_vel command during the calculations of the new path, which makes the robot halts for a small amount of time. 

So my question is, could I publish my goals to a topic as via points, to make the robot follow the path?

Otherwise, my plan is to do a watchdog that looks at the cmd_vel topic and as soon as it stops (due to new calculation of path) it continues sending the previous cmd_vel command. What do you think @croesmann ?