rst-tu-dortmund/teb_local_planner #126: Footprint model error

hs2041, 2019-04-16:
When I run move_base with teb configured, I get the following warning:
[ WARN] [1555451338.153400931, 6561.944000000]: The inscribed radius of the footprint specified for TEB optimization (0.200000) + min_obstacle_dist (0.750000) are smaller than the inscribed radius of the robot's footprint in the costmap parameters (179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, including 'footprint_padding'). Infeasible optimziation results might occur frequently!
[ WARN] [1555451339.005904036, 6562.279000000]: Map update loop missed its desired rate of 4.0000Hz... the loop actually took 0.4660 seconds

croesmann, 2019-06-28:
I guess that your costmap_2d footprint is not configured properly. Can you post your costmap_2d parameters here?

alexbaucom, 2019-12-17:
I am having a similar error. It seems like it may be crashing TEB because I can't get move base to work now.

My costmap footprint parameter is:
`footprint: [[0.2705, 0.245], [ -0.2705, 0.245], [-0.365, 0.0], [-0.2705, -0.245], [0.2705, -0.245], [0.365, 0.0]]`

And with a little bit of padding added (I think just the default of 0.01), I can see it being published on `/move_base/local_planner/footprint` as
```
header: 
  seq: 425
  stamp: 
    secs: 1576618088
    nsecs: 301387702
  frame_id: "odom"
polygon: 
  points: 
    - 
      x: 0.280355036259
      y: 0.255159378052
      z: 0.0
    - 
      x: -0.280644863844
      y: 0.254840552807
      z: 0.0
    - 
      x: -0.374999940395
      y: -0.000213108651224
      z: 0.0
    - 
      x: -0.280355036259
      y: -0.255159378052
      z: 0.0
    - 
      x: 0.280644863844
      y: -0.254840552807
      z: 0.0
    - 
      x: 0.374999940395
      y: 0.000213108651224
      z: 0.0
```
Note that everything seems to work fine with DWA planner. For some reason only TEB seems to be unhappy about the footprint.

Any help you can provide would be appreciated!

alexbaucom, 2019-12-26:
I did some more digging here and I believe the issue shows up when using an older version of the navigation package. At least in my case it seems that the older navigation package my organization is using (with some of our own custom stuff on top) has some older function prototypes that the newest versions of TEB aren't compatible with. 

When I installed TEB via apt, it would look like it was running okay, but then return this warning when the `costmap_ros` pointer it was given was not what it expected (which in turn caused it to get an empty footprint, and then finally throw up this warning when trying to calidate min/max distances on the empty footprint).

Once I actually attempted to build TEB from source against the customized nav stack, I found that the compilation would fail due to a few mismatched functions in `costmap_2d_ros`, namely calls to `costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&)` and `base_local_planner::OdometryHelperRos::getRobotVel(geometry_msgs::PoseStamped&)` which both take in `tf::Stamped<tf::Pose>&` values in the version of the nav code that I have.

closed 2020-06-10 by hs2041