rst-tu-dortmund/teb_local_planner #131: rviz crash while using test_optim_node

coreZair, 2019-04-24:
Hi Devs,

RViz keeps crashing while trying out the default example (test_optim_node). Tried out in Ubuntu Xenial 16.04 LTS on kinetic-dev branch and Ubuntu Bionic 18.04 LTS on melodic-dev branch.
Most times RViz works for some mouse clicks, handling the InteractiveMarkers, setting via points with "Publish Point" Button. But after some clicks RViz crashes ("process has died"). When you keep the roscore running and reopen RViz after the crash again, then clicking on "Publish Point" or Left Click on an InteractiveMarker is enough to make RViz crash immediately. After restarting the roscore RViz stays stable
Everything is up to date. Latest ROS and kernel installed via apt. RViz crashes when your test_optim_node is running only.

Are there any known issues guys? Or do you have any other example node to cross check if it has something to do with teb_local_planner itself or RViz only?
Greetings from Dresden.

Kind regards,
Patrick

coreZair, 2019-04-25:
I found an answer in the RViz issue tracker! It has nothing to do with the teb_local_planner itself, it is related to RViz or NVidia drivers. I don't know which of these both cause the crash, but using a NVidia graphics card with proprietary drivers crashes RViz, while using InteractiveMarkers AND MarkerArrays with the types LINE_STRIP or LINE_LIST.

Workaround: Change Marker types with LINE_LIST or LINE_STRIP to CUBE_LIST, POINTS, or SPHERE_LIST. Or simply deactivate the InteractiveMarkers in RViz. Another way around is to use the Processor's integrated graphic unit (f.e. in Intel processors). Maybe an open source nvidia driver like "nouveau".

Kind regards,
Patrick

closed 2019-04-25 by coreZair

guannan-he, 2021-06-18:
Deactive "Marker->Namespaces->TebContainer" before publish a point will also work, reactive the same option to visualize alternative trajectories if necessary.