rst-tu-dortmund/teb_local_planner #132: Using TEB Local Planner with Move Base Flex

amrp1, 2019-05-14:
Thank you @croesmann for this great controller. 

I would like to use TEB Local Planner with [Move Base Flex](http://wiki.ros.org/move_base_flex) to request the Planner for a new plan when the current path is blocked. 

Currently I am giving a set of waypoints that are being executed by calling the [ExePathAction](http://docs.ros.org/kinetic/api/mbf_msgs/html/action/ExePath.html) with TEBLocalPlannerROS as the 'controller'. When the controller encounters an obstacle, it tends to ask for a new plan from the planner. I want to use this event as a trigger to call the [GetPathAction](http://docs.ros.org/kinetic/api/mbf_msgs/html/action/GetPath.html), through which I can use my desired plan to issue a new plan. For this, I need the TEB Local Planner to return " BLOCKED_PATH = 109"   as the result in the ExePathAction.

Which section of the controller code governs the sudden appearance of obstacles on the path? Is it possible to generate the BLOCKED_PATH outcome?

Thank you for your time :+1:

croesmann, 2019-06-28:
hi @amrp1,

I haven't worked with MBF yet. But @corot told me that he is going to provide a PR for MBF support soon :-)

corot, 2019-06-28:
Hi all,
yes, right, we made TEB a MBF plugin [here](https://github.com/magazino/teb_local_planner/tree/mbf_plugin). We will PR into upstream next week. But this won't give u the BLOCKED_PATH outcome, unless you modify TEB accordingly. Our plugin doesn't return BLOCKED_PATH cause this condition is not detected on current implementation. The closest it gets is the isTrajectoryFeasible method, but this is actually checking the trajectory generated by TEB, not if the global path provided by the global planner is blocked, what I suppose is what you want. I see 2 possible solutions:

- Modify TEB to do this check, e.g. reusing [this loop](https://github.com/rst-tu-dortmund/teb_local_planner/blob/kinetic-devel/src/teb_local_planner_ros.cpp#L695) (we do that in our version of DWA)

- Call _check_path_cost_ service on MBF before calling TEB. This will give you all you need, as the service [request/result](https://github.com/magazino/move_base_flex/blob/master/mbf_msgs/srv/CheckPath.srv) are very  detailed (please open an issue in MBF if you think something is missing or not clear enough)

amrp1, 2019-07-04:
Thank you for the suggestions, @corot. Will try them out. Closing this issue for now.

closed 2019-07-04 by amrp1