rst-tu-dortmund/teb_local_planner #136: Error during source code compilation

nature1949, 2019-05-28:
When I did the source compilation process, I get an error:
`/home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp: In member function ‘virtual bool teb_local_planner::TebLocalPlannerROS::computeVelocityCommands(geometry_msgs::Twist&)’:
/home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:229:40: error: no matching function for call to ‘costmap_2d::Costmap2DROS::getRobotPose(tf::Stamped<tf::Transform>&)’
   costmap_ros_->getRobotPose(robot_pose);
                                        ^
In file included from /home/wc/catkin_ws/src/basic/navigation/nav_core/include/nav_core/base_local_planner.h:42:0,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:45,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/wc/catkin_ws/src/basic/navigation/costmap_2d/include/costmap_2d/costmap_2d_ros.h:125:8: note: candidate: bool costmap_2d::Costmap2DROS::getRobotPose(geometry_msgs::PoseStamped&) const
   bool getRobotPose(geometry_msgs::PoseStamped& global_pose) const;
        ^
/home/wc/catkin_ws/src/basic/navigation/costmap_2d/include/costmap_2d/costmap_2d_ros.h:125:8: note:   no known conversion for argument 1 from ‘tf::Stamped<tf::Transform>’ to ‘geometry_msgs::PoseStamped& {aka geometry_msgs::PoseStamped_<std::allocator<void> >&}’
/home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:234:40: error: no matching function for call to ‘base_local_planner::OdometryHelperRos::getRobotVel(tf::Stamped<tf::Transform>&)’
   odom_helper_.getRobotVel(robot_vel_tf);
                                        ^
In file included from /home/wc/catkin_ws/src/basic/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:47:0,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/wc/catkin_ws/src/basic/navigation/base_local_planner/include/base_local_planner/odometry_helper_ros.h:66:8: note: candidate: void base_local_planner::OdometryHelperRos::getRobotVel(geometry_msgs::PoseStamped&)
   void getRobotVel(geometry_msgs::PoseStamped& robot_vel);
        ^
/home/wc/catkin_ws/src/basic/navigation/base_local_planner/include/base_local_planner/odometry_helper_ros.h:66:8: note:   no known conversion for argument 1 from ‘tf::Stamped<tf::Transform>’ to ‘geometry_msgs::PoseStamped& {aka geometry_msgs::PoseStamped_<std::allocator<void> >&}’
In file included from /opt/ros/kinetic/include/class_loader/class_loader_core.hpp:46:0,
                 from /opt/ros/kinetic/include/class_loader/class_loader.hpp:43,
                 from /opt/ros/kinetic/include/class_loader/./multi_library_class_loader.hpp:42,
                 from /opt/ros/kinetic/include/class_loader/multi_library_class_loader.h:35,
                 from /opt/ros/kinetic/include/pluginlib/class_loader.hpp:38,
                 from /home/wc/catkin_ws/src/basic/navigation/costmap_2d/include/costmap_2d/costmap_2d_ros.h:50,
                 from /home/wc/catkin_ws/src/basic/navigation/nav_core/include/nav_core/base_local_planner.h:42,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:45,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/opt/ros/kinetic/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::class_loader_private::MetaObject<C, B>::create() const [with C = teb_local_planner::TebLocalPlannerROS; B = nav_core::BaseLocalPlanner]’:
/home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:1159:1:   required from here
/opt/ros/kinetic/include/class_loader/meta_object.hpp:198:16: error: invalid new-expression of abstract class type ‘teb_local_planner::TebLocalPlannerROS’
     return new C;
                ^
In file included from /home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:39:0:
/home/wc/catkin_ws/src/basic/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:92:7: note:   because the following virtual functions are pure within ‘teb_local_planner::TebLocalPlannerROS’:
 class TebLocalPlannerROS : public nav_core::BaseLocalPlanner
       ^
In file included from /home/wc/catkin_ws/src/basic/teb_local_planner/include/teb_local_planner/teb_local_planner_ros.h:45:0,
                 from /home/wc/catkin_ws/src/basic/teb_local_planner/src/teb_local_planner_ros.cpp:39:
/home/wc/catkin_ws/src/basic/navigation/nav_core/include/nav_core/base_local_planner.h:78:20: note: 	virtual void nav_core::BaseLocalPlanner::initialize(std::__cxx11::string, tf2_ros::Buffer*, costmap_2d::Costmap2DROS*)
       virtual void initialize(std::string name, tf2_ros::Buffer* tf, costmap_2d::Costmap2DROS* costmap_ros) = 0;
                    ^
`
What is the cause of the error? And what should I do?

minghuizhou123, 2019-06-03:
me too, Have you solved it?

minghuizhou123, 2019-06-04:
[image]

YujieLu, 2019-06-12:
Maybe you should check out the branch according to the distribution of your ros. For me, I checkout melodic-devel, then all problems solved.

TanyuSir, 2019-07-29:
Did you solve this error? I have encountered the same problem.

TanyuSir, 2019-07-29:
I have encountered the same problem.But I solved it by the following method.
in your catkin_ws
git clone https://github.com/rst-tu-dortmund/costmap_converter.git
then
catkin_make

TanyuSir, 2019-07-31:
The above problem is solved by the following methods
https://answers.ros.org/question/329601/teb_local_planner-compilation-error/

closed 2019-07-31 by croesmann