rst-tu-dortmund/teb_local_planner #153: complete_global_plan not listed on wiki

rittersb, 2019-06-27:
I noticed that `complete_global_plan` is not listed on the [wiki](http://wiki.ros.org/teb_local_planner) page, and I only found it when I started to dig around in the code. It is read in as a param in [teb_config.cpp](https://github.com/rst-tu-dortmund/teb_local_planner/blob/6db7c2d6071f543948c388f175e4563e6ffbc909/src/teb_config.cpp#L87). I had an issue where `goal_reached_` would not return true, and it was due to the check done in [teb_local_planner_ros.cpp](https://github.com/rst-tu-dortmund/teb_local_planner/blob/6db7c2d6071f543948c388f175e4563e6ffbc909/src/teb_local_planner_ros.cpp#L266), and setting `complete_global_plan` to false helped resolve that issue.

croesmann, 2019-06-28:
you are right, the parameter is not listed on the wiki page.
It was introduced in #43. However, your issue made me rethink the function. Maybe your use case is not considered correctly in the code. I guess you are using via points since otherwise `via_points_.size() == 0` would render `complete_global_plan` obsolete.

rittersb, 2019-06-28:
I was utilizing `global_plan_viapoint_sep` so that it would generate the via points for me. Here is my config file if you are interested. It has been a while since I looked at this, but from what I remember I would give the robot a goal, but it would never quite reach it even though it was close enough to the goal due to the `xy_goal_tolerance` and the `yaw_goal_tolerance`. I would only get the case where `via_points.size() != 0` hence it never reaching the goal.
```
TebLocalPlannerROS:
  #Robot Configuration
  acc_lim_x: 1.0
  acc_lim_theta: 1.0
  max_vel_x: 1.0
  max_vel_y: 0.0
  acc_lim_y: 0.0
  max_vel_x_backwards: 0.05
  max_vel_theta: 0.05
  min_turning_radius: 0.0
  # footprint_model/type: line
  # footprint_model/line_start: [-0.50,0.0]
  # footprint_model/line_end: [1.25,0.0]
  footprint_model/type: polygon
  footprint_model/vertices: [[-0.75, -0.8], [-0.75, 0.8], [1.0, 0.8], [1.0, -0.8]]
  # footprint_model/type: two_circles
  # footprint_model/front_offset:
  # footprint_model/rear_offset:
  # footprint_model/front_radius: 0.635
  # footprint_model/rear_radius: 0.635
  # #Goal Tolerance
  xy_goal_tolerance: 1.0
  yaw_goal_tolerance: 6.2831853072
  complete_global_plan: false
  free_goal_vel: true
  # Trajectory Configuration
  dt_ref: 0.3
  dt_hysteresis: 0.1
  max_global_plan_lookahead_dist: 10.0
  global_plan_overwrite_orientation: false
  global_plan_viapoint_sep: 0.75
  # Obstacle Parameters
  min_obtacle_dist: 0.7
  inflation_dist: 1.0
  costmap_obstacles_behind_robot_dist: 1.0
  obstacle_poses_affected: 15
  # Optimization Parameters
  penalty_epsilon: 0.001
  no_inner_iterations: 5
  no_outer_iterations: 4
  weight_kinematics_forward_drive: 1000.0
  weight_max_vel_x: 2
  weight_max_vel_theta: 1
  weight_acc_lim_x: 100
  weight_acc_lim_theta: 100
  weight_kinematics_turning_radius: 0.0
  weight_viapoint: 75
  # Parallel Planning
  selection_cost_hysteresis: 1.5

  # Miscellaneous
  odom_topic: /my_bot/mobile_base_controller/odom
  map_frame: /odom
```