rst-tu-dortmund/teb_local_planner #154: ROS 2 Integration Plans

yathartha3, 2019-06-27:
The teb_local_planner for ROS1 is very useful for the robotics community, and I am curious if you have any plans for porting it to ROS2.
I am interested in helping in the porting process, if required, or try to port it myself. However, since you are the experts on this, I would like to know what your plans are :)

croesmann, 2019-06-28:
Yes, porting the planner to ROS2 is definitely on the list.
However, I wanted to wait until ROS2 and navigation2 are more stable, so I haven't checked any progress/updates in the last few months.
But I just had a look at the navigation2 repo and it seems to be in a good state.
Also in https://github.com/ros-planning/navigation2/issues/202 a certain interest in testing the teb_local_planner with the new interface is mentioned.
However, I have not yet started to review the best practices for the Navigation2 codebase and ROS2 migration. 
A very important question comes to my mind:
How do we handle maintenance and new feature development for both ROS1 and ROS2?
Just duplicating the package to teb_local_planner2 would increase the effort substantially.
Another way would be to divide the current package into several subpackages/modules so that both ROS1 and ROS2 wrappers can share the same planner backend. However, this would also imply substantial changes to teb_local_planner1. Are you aware of any best practices? Probably, the same problem applies to many other ROS packages. I'll be investigating that soon.

crdelsey, 2019-06-28:
> How do we handle maintenance and new feature development for both ROS1 and ROS2?

The most common solution I've seen so far is to keep the ROS 2 version on a separate branch. That way, if you make a change in ROS 1, you can just merge it to the ROS 2 version. It's still going to be extra work because you'll frequently have merge conflicts, but at least Git will help you keep track of what they are.

> navigation2 repo and it seems to be in a good state

I believe we're getting close anyway :-)

teb_local_planner seemed like a very different approach to local planning than DWA, so it's a good test of the architecture. I'd like to fix any architecture problems in navigation2 that make it harder to implement teb than it should be.

croesmann, 2019-07-16:
many thanks, @crdelsey, for providing me with the information. A ros2 branch seems to be best suited at the moment. 

Crucial dependencies are:
- costmap_converter
I'll release costmap_converter ([this PR](https://github.com/rst-tu-dortmund/costmap_converter/pull/10)) to ros2 as soon as navigation2 is in dashing (which should be happening soon according to [rosdistro](https://github.com/ros/rosdistro/pull/21764)). 
- base_local_planner
 Already noticed [here (navigation2)](https://github.com/ros-planning/navigation2/issues/931)
- libg2o
 This package is just a wrapper to the original cmake g2o package. I need to check whether this procedure applies to ros2 best practices, too. I started a discussion [here](https://github.com/ros-gbp/libg2o-release/issues/9).

croesmann, 2019-07-16:
To my mind, we might also remove the dependency on base_local_planner, as the teb_local_planner
only uses a few utility functions from base_local_planner that could be reimplemented.

SteveMacenski, 2019-07-16:
We may want to consider an extended planner API for pluginlib before trying to add more external planners. This is the easiest it’s ever going to be to decide to add new methods for contextual planner behaviors or extending. An example would be David’s nav_core2.

vinnamkim, 2019-07-21:
@yathartha3 @crdelsey @croesmann @SteveMacenski 
Hi guys, I have some works porting TEB planner to ROS2 at [my forked repo](https://github.com/vinnamkim/teb_local_planner/tree/dashing-devel). It is integrated with navigation2(Like DWB planner of navigation2) **without base_local_planner**. I have a plan to PR this to upstream. I guess we can use these works. However, some instability of navigation2 makes me hard to synchronize with the latest change of navigation2. I'll try to make it working with the latest navigation2 as soon as possible and make pull request.

yathartha3, 2019-07-22:
@vinnamkim awesome!
I currently have a ported base_local_planner, and am working on defining an interface with common functionalities that will work for any local planner. We are still working on defining the details for interface. You should still submit a PR for your current TEB, and we can refactor it later.

yathartha3, 2019-09-21:
ROS2 navigation now has plugin interface for local planners. This allows TEB to be used as plugin.
@vinnamkim, since you ported it, so you want to go ahead and make it a plugin?
We currently have DWB working as plugin too. So having TEB as plugin will be awesome!
I will be happy to collaborate.

SteveMacenski, 2019-09-21:
@vinnamkim I can also commit to helping you port as needed and understand your needs to integrate into the larger navigation ecosystem if need be. 

The ROS2 navigation plugin definitions have been made in `nav2_core` and we'd love to see TEB be one of the first ROS2 ported options. Down the line we're looking at multiple controller and planner options running in parallel for specific purposes (if A fails, go to B, if B fails, go to recovery type logic or BT enable contextual controllers).

vinnamkim, 2019-09-23:
@yathartha3 @SteveMacenski Thanks for notifications, I'll start to follow up with the plugin interface in `nav2_core` and make pull request to update TEB as plugin as soon as possible I can.

EvenRiver, 2019-11-20:
> @yathartha3 @SteveMacenski Thanks for notifications, I'll start to follow up with the plugin interface in nav2_core and make pull request to update TEB as plugin as soon as possible I can.

@vinnamkim I have failed to build the latest dashing-devel code, it seemed the dependency problem, there is no corresponding libg2o package for ros2. Can you give me some help?

croesmann, 2019-11-20:
Currently, I am facing an issue with releasing libg2o as third party package into ros2 (see [here](https://github.com/ament/ament_tools/issues/163#issuecomment-548103755)).

Workaround for testing: You can install g2o manually and it is likely to be found by the findG2O.cmake script.

EvenRiver, 2019-11-21:
Thanks @croesmann , I have manually installed libg2o as you said, but then met some new problems as below. 

```
    --- stderr: teb_local_planner                               
In file included from /home/even/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:665:0,
                 from /home/even/navigation2_ws/install/nav2_util/include/nav2_util/lifecycle_node.hpp:23,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/include/teb_local_planner/teb_config.h:42,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/include/teb_local_planner/visualization.h:45,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/src/visualization.cpp:40:
/home/even/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:26:10: fatal error: rclcpp/experimental/intra_process_manager.hpp: No such file or directory
 #include "rclcpp/experimental/intra_process_manager.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/even/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:665:0,
                 from /home/even/navigation2_ws/install/nav2_util/include/nav2_util/lifecycle_node.hpp:23,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/include/teb_local_planner/teb_config.h:42,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/include/teb_local_planner/optimal_planner.h:46,
                 from /home/even/teb_ws/src/teb_local_planner-dashing-devel/teb_local_planner/src/optimal_planner.cpp:42:
/home/even/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:26:10: fatal error: rclcpp/experimental/intra_process_manager.hpp: No such file or directory
 #include "rclcpp/experimental/intra_process_manager.hpp"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```
I noticed that the path for "intra_process_manager.hpp" has changed from "rclcpp/intra_process_manager.hpp" to  "rclcpp/experimental/intra_process_manager.hpp" since ROS2 Eloquent. So this is may be a compatible problem, can you have a check about this?

EvenRiver, 2019-11-22:
hi @croesmann , I got the it done. When I installed ROS2 from source, the errors appeared, but it disappeared once I installed ROS2 from Debian packages

maxlein, 2020-01-03:
Is there already a release for eloquent in the making?  
I also tried building from source, and fixing some of the seg faults and errors, but as I am not fully aware of the code, I get myself into more trouble I think.

jplapp, 2020-01-07:
We have investigated this recently - I have created #187 to adapt teb for the nav2 eloquent release.

SteveMacenski, 2020-01-22:
I think this ticket can be safely closed now with the successful dashing release and the updates for eloquent in that PR.

We have successfully planned and executed :wink:

croesmann, 2020-01-23:
I am currently working on the release ;-). Dependencies are already on the way to the buildfarm,
at least in eloquent. However, I cannot release into dashing atm, as the released navigation2 package does not provide the `nav_core` module. Am I missing something @SteveMacenski ?

SteveMacenski, 2020-01-23:
its `nav2_core`, can't tell if that's a typo here or in your package.xml. 

It is true, however, that there is no nav2_core or plugin interface for Dashing. Only Eloquent and newer.

croesmann, 2020-01-23:
yes, I meant `nav2_core`.
Okay, so I'll revert #174 in dashing-devel and just keep those changes for eloquent.

SteveMacenski, 2020-01-23:
I think you could just entirely remove the dashing branch. Functionally, it doesn't exist if there's no interface to go to (unless beforehand you had your own TEB server?).

Dashing's in a weird state for alot of projects including Navigation. I would not recommend anyone use Dashing for many reasons, least of which are the major stability increases in Eloquent and far more maturity in Navigation

SteveMacenski, 2020-10-09:
@croesmann  I think this can be closed if you just run bloom on Foxy!

closed 2021-03-08 by SteveMacenski