rst-tu-dortmund/teb_local_planner #161: move_base not working since using teb instead of base_local_planner

Daniel2401, 2019-07-29:
Hi,

after switching from base_local_planner to teb I get the following error message every time I give a new 2D navigation goal through rviz (similar to #125. If my new Issue is unwanted I can comment under #125, but I thought this might be a bit long for a comment in a somewhat dead Issue).

```
/home/moro01/catkin_ws/devel/lib/move_base/move_base: symbol lookup error: /opt/ros/kinetic/lib//libteb_local_planner.so: undefined symbol: _ZNK10costmap_2d12Costmap2DROS12getRobotPoseERN2tf7StampedINS1_9TransformEEE
[move_base-3] process has died [pid 4897, exit code 127, cmd /home/moro01/catkin_ws/devel/lib/move_base/move_base __name:=move_base __log:=/home/moro01/.ros/log/332d6ccc-b246-11e9-82f4-94c691a5aba3/move_base-3.log].
log file: /home/moro01/.ros/log/332d6ccc-b246-11e9-82f4-94c691a5aba3/move_base-3*.log

```

One warning that didn't show up with base_local_planner when launching move_base.launch is  

```
[ WARN] [1564435799.756196141]: The inscribed radius of the footprint specified for TEB optimization (0.000000) + min_obstacle_dist (5.000000) are smaller than the inscribed radius of the robot's footprint in the costmap parameters (179769313486231570814527423731704356798070567525844996598917476803157260780028538760589558632766878171540458953514382464234321326889464182768467546703537516986049910576551282076245490090389328944075868508455133942304583236903222948165808559332123348274797826204144723168738177180919299881250404026184124858368.000000, including 'footprint_padding'). Infeasible optimziation results might occur frequently!
```
I have yet to find a solution since no changes in either my teb file or my costmap_common_params file fixed this warning.

My robot's footprint is 0.65 m x 0.485 m so I want to use the line footprint. If I want 0.20 m distance between the robot and obstacles my min_obstacle_dist has to be 0.485/2 + 0.20 = 0.4425 and the line ends at +-0.0825 (0.65/2 - 0.485/2 ) right?

Here are my files:

move_base.launch
```
<launch>

   <master auto="start"/>
   <!--master auto="start"/-->
 <!-- Run the map server --> 
    <!--node name="map_server" pkg="map_server" type="map_server" args="$(find moro_2dnav)/mymap03.pgm 0.05"/-->
    <node name="map_server" pkg="map_server" type="map_server" args="$(find moro_2dnav)/buero_georg_map.pgm 0.050000"/>

 <!--- Run AMCL --> 
    <include file="$(find amcl)/examples/amcl_diff.launch" />

   <!--node pkg="move_base" type="move_base" respawn="true" name="move_base" output="screen"-->
   <node pkg="move_base" type="move_base" respawn="false" name="move_base" output="screen">
    <rosparam file="$(find moro_2dnav)/costmap_common_params.yaml" command="load" ns="global_costmap" /> 
    <rosparam file="$(find moro_2dnav)/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find moro_2dnav)/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find moro_2dnav)/global_costmap_params.yaml" command="load" /> 
    <rosparam file="$(find moro_2dnav)/base_local_planner_params.yaml" command="load" />

    <param name="base_local_planner" value="teb_local_planner/TebLocalPlannerROS" />
    <param name="controller_frequency" value="10.0" />

 </node>

</launch> 
```

base_local_planner_params.yaml (teb file)
```
TebLocalPlannerROS:

 odom_topic: odom
 map_frame: /odom
    
 # Trajectory
  
 teb_autosize: True
 dt_ref: 0.3
 dt_hysteresis: 0.1
 global_plan_overwrite_orientation: True
 max_global_plan_lookahead_dist: 3.0
 feasibility_check_no_poses: 5
    
 # Robot
         
 max_vel_x: 0.4
 max_vel_x_backwards: 0.2
 max_vel_theta: 0.3
 acc_lim_x: 0.5
 acc_lim_theta: 0.5
 min_turning_radius: 0.0
 footprint_model: # types: "point", "circular", "two_circles", "line", "polygon"
   type: "line"
   radius: 0.2 # for type "circular"
   line_start: [-0.0825, 0.0] # for type "line"
   line_end: [0.0825, 0.0] # for type "line"
   front_offset: 0.2 # for type "two_circles"
   front_radius: 0.2 # for type "two_circles"
   rear_offset: 0.2 # for type "two_circles"
   rear_radius: 0.2 # for type "two_circles"
   vertices: [ [0.25, -0.05], [0.18, -0.05], [0.18, -0.18], [-0.19, -0.18], [-0.25, 0], [-0.19, 0.18], [0.18, 0.18], [0.18, 0.05], [0.25, 0.05] ] # for type "polygon"

 # GoalTolerance
    
 xy_goal_tolerance: 0.2
 yaw_goal_tolerance: 0.1
 free_goal_vel: False
    
 # Obstacles
    
 min_obstacle_dist: 0.4425
 include_costmap_obstacles: True
 costmap_obstacles_behind_robot_dist: 1.0
 obstacle_poses_affected: 30
 costmap_converter_plugin: ""
 costmap_converter_spin_thread: True
 costmap_converter_rate: 5

 # Optimization
    
 no_inner_iterations: 5
 no_outer_iterations: 4
 optimization_activate: True
 optimization_verbose: False
 penalty_epsilon: 0.1
 weight_max_vel_x: 2
 weight_max_vel_theta: 1
 weight_acc_lim_x: 1
 weight_acc_lim_theta: 1
 weight_kinematics_nh: 1000
 weight_kinematics_forward_drive: 1
 weight_kinematics_turning_radius: 1
 weight_optimaltime: 1
 weight_obstacle: 50
 weight_dynamic_obstacle: 10 # not in use yet
 alternative_time_cost: False # not in use yet

 # Homotopy Class Planner

 enable_homotopy_class_planning: True
 enable_multithreading: True
 simple_exploration: False
 max_number_classes: 4
 roadmap_graph_no_samples: 15
 roadmap_graph_area_width: 5
 h_signature_prescaler: 0.5
 h_signature_threshold: 0.1
 obstacle_keypoint_offset: 0.1
 obstacle_heading_threshold: 0.45
 visualize_hc_graph: False
```

costmap_common_params.yaml
```
obstacle_range: 2.5
raytrace_range: 3.0
footprint: [[-0.325, -0.2425], [-0.325, 0.2425], [0.325, 0.2425], [0.325, -0.2425]]
#robot_radius: ir_of_robot
inflation_radius: 0.62

observation_sources: laser_scan_sensor

laser_scan_sensor: {sensor_frame: base_laser, data_type: LaserScan, topic: scan, marking: true, clearing: true}
```

simonbogh, 2019-08-08:
Hi,

I am no expert in TEB (yet), but it seems you have configured the robot `footprint` as a box/polygon in costmap_common_params.yaml, but in base_local_planner_params.yaml (teb file) you define the `footprint_model` as "line". In TEB, try and define your robot as "polygon" with the same values as in costmap_common_params.yaml. 

Alternatively you can try and set both to a circle, just for testing. That would be `robot_radius` in costmap_common_params.yaml, and `type: "circular"`, and set the same radius.

croesmann, 2019-08-09:
The linker error is probably due to different versions for move_base and the teb_local_planner.
Are you mixing kinetic and melodic?
Refer to [this question and answers](https://answers.ros.org/question/329601/teb_local_planner-compilation-error/).

Daniel2401, 2019-08-29:
Hi,

yes it seams as the issue was due to different versions, although I used kinetic for both and both should have been up to date.  Building move_base from src fixed it anyway.

Thank you for your help.

closed 2019-08-29 by Daniel2401