rst-tu-dortmund/teb_local_planner #163: Which robot footprint model type requires the least computation time?

WLwind, 2019-08-02:
It says in wiki that "The type of the model significantly influences the required computation time.".
So I wonder which one requires the least computation time?

simonbogh, 2019-08-08:
I do not know exactly for TEB, but it would usually be circular footprints as the robot can be treated as a point and easily compared to the inflated costmap. Complex polygon shaped footprints usually take more computational power.

A bit old, but this paper may be an interesting read:
**Efficient Cost Computation in Cost Map Planning for Non-Circular Robots**
http://www.cs.cmu.edu/~maxim/files/footprintoptim_iros09.pdf

croesmann, 2019-08-09:
@simonbogh already provided the correct answer.
Point or circular footprints require the least computation time.

The two-circles model requires two distance calculations.
The line model includes some additional simple case checking, however, the computation time should be still okay for most applications.
The polygon model requires a distance check for each edge and hence computation time significantly depends on the number of edges/vertices.

See also [here](http://wiki.ros.org/teb_local_planner/Tutorials/Obstacle%20Avoidance%20and%20Robot%20Footprint%20Model)

closed 2019-08-09 by croesmann