rst-tu-dortmund/teb_local_planner #165: Determining the max translational and angular acceleration ?

gsh89, 2019-08-16:
If the max translational and angular acceleration are not available, what is a good way to obtain these values for a given robot?

croesmann, 2019-08-16:
Hi, 
this question is better placed at answers.ros.org.
However, consider the following options:
- If you know your maximum motor/wheel speed, you can compute them using simple kinematics, e.g. for a differential drive robot. You can find the equations in every robotics textbook.
- If you have a dynamic model of your robot, you can try to obtain them by simulation
- Perform some tests with your real robot: increase translational and rotational velocities systematically and validate the actual motion.

gsh89, 2019-08-16:
@croesmann Thank you for the pointers.

closed 2019-08-16 by gsh89